WebODM/WebODM #446: Remove reset password link

pierotofy, 2018-05-07:
Users are getting confused because the "reset password" link actually cannot send e-mails. Better to instruct them on how to reset the password via terminal.

closed 2018-05-13 by pierotofy

Bornlove, 2021-09-19:
Hello @pierotofy 
have tried to reset password but when trying to log in it still rejects, does this cmd ./webodm.sh resetadminpassword yournewpass changes Username as well?

pierotofy, 2021-09-19:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

Bornlove, 2021-09-20:
> Could we move this conversation over to the forum at https://community.opendronemap.org? 🙏 The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! 👍

Thanks for kindly reply @pierotofy. Appreciate

PR #451 (2018-05-13): Improvements
Various improvements.

Fixes #446 #445