WebODM/WebODM #452: Hang during Uploading images

DelphiR, 2018-05-14:
How did you install WebODM? (Docker, natively, ...)?
Docker:
git clone https://github.com/OpenDroneMap/WebODM
cd WebODM
./webodm.sh start (with 443 port and certificate)

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome 66 64-bit

What is the problem?
Sometimes the task is hanging at the state "Uploading images". This state will be "Uploading images" for constant time and never disappeared. I checked such methods as:
- restart node
- restart computer with docker running 
- restart network connection
Noone didn't work.
Only removing and adding task again help.

DelphiR, 2018-05-14:
Is there any way to check how much timeout now? Maybe I just didn'nt wait enough?
I think that it was due to became offline of some node and then go back it to online.
That seems that there is no way to change another node automatically if smth go wrong.

pierotofy, 2018-05-15:
The application has very relaxed timeout values (currently set at 24 hours), so unless your dataset took longer than that, it shouldn't have been a problem with the timeout settings. Perhaps the connection was dropped because of some other issue. Could you share some more information about your servers' infrastructure?

pierotofy, 2019-01-15:
@DelphiR were you able to figure out the cause of the hanging?

pierotofy, 2021-02-03:
Closing due to inactivity; this has likely improved since 2018.

closed 2021-02-03 by pierotofy