WebODM/WebODM #453: Hang during running processing

DelphiR, 2018-05-14:
How did you install WebODM? (Docker, natively, ...)?
Docker:
git clone https://github.com/OpenDroneMap/WebODM
cd WebODM
./webodm.sh start (with 443 port and certificate)

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome 66 64-bit

What is the problem?
I have made an attempt to use several nodes. I noticed the several cases when the running status became continous and never end. I think that the lack of timeout and control of nodes cause the such errors. I believe that the right decision will be support for timeouts and choose another node to tru to run on it and so on.

pierotofy, 2018-05-15:
Is it possible that your server ran out of memory while processing? How much RAM do you have?

DelphiR, 2018-05-16:
I think yes. I use 2 servers for testing - with 8 and 16 gb of RAM
I believe that there are a lot of possible problems. But there might be a logic when service send the task to another node when using auto mode just to try. Or restart, or smth else