WebODM/WebODM #454: Potree incorrect measurements

DelphiR, 2018-05-14:
How did you install WebODM? (Docker, natively, ...)?
Docker:
git clone https://github.com/OpenDroneMap/WebODM
cd WebODM
./webodm.sh start (with 443 port and certificate)

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome 66 64-bit

What is the problem?
It seems that Potree editor which display the points cloud give incorrect measurements.
The last docker seems to be used the 1.5 of Potree.
There is no any info about howto to use this editor so the situation near Webodm project is not good for newbiee due to lack of documentation.
The error looks like:
https://ibb.co/cN77Vd

DelphiR, 2018-05-14:
Here there is a discussion which means that the old version of potree may contains an error
https://github.com/potree/potree/issues/403
Please could you renew Webodm to use newer version of Potree?

DelphiR, 2018-05-14:
IT seems that square measurement just measure smth else. Maybe the error in coordinate system? The sqaure to small to be calculated as the XY projection.

pierotofy, 2018-05-14:
Thanks for reporting the problem! It looks like an issue with Potree and updating to the latest version should fix it.

DelphiR, 2018-05-14:
OK. Please take a look and get us info how can we update it on docker or docker after WebODM commit. The Potree and WebODM are great together. The square measure and volume are main and important.

pierotofy, 2019-06-08:
Hey @DelphiR, we've just released an updated version which upgrades the version of Potree. Could you check if this issue has been resolved?

closed 2020-04-02 by pierotofy