WebODM/WebODM #456: Processing fails when GCP file added issue with python?

DChum17, 2018-05-23:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 60 on Windows 10

### What is the problem?

Data set runs fine when no GCP file loaded but fails when one is added

### What should be the expected behavior?

Webodm finishes with no problems

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

This is the output from webodm with no GCP

[INFO]    Found 23 usable images
[DEBUG]   Line: WGS84 UTM 53N
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   Altitude data detected, enabling it for GPS alignment
[DEBUG]   ['use_exif_size: no', 'feature_process_size: 2048', 'feature_min_frames: 12000', 'processes: 4', 'matching_gps_neighbors: 12', 'optimize_camera_parameters: yes', 'use_altitude_tag: True', 'align_method: naive']
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /var/www/data/2f396480-fd4d-4020-9db4-6cc972d39c8f/opensfm

The output with GCP

[INFO]    Found 23 usable images
[DEBUG]   Line: EPSG:6672 JP2011 JP
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/dataset.py", line 86, in process
outputs.reconstruction = types.ODM_Reconstruction(photos, coords_file=tree.odm_georeferencing_gcp)
File "/code/opendm/types.py", line 140, in __init__
self.projection = self.parse_coordinate_system(coords_file)
File "/code/opendm/types.py", line 168, in parse_coordinate_system
return Proj(init=line)
File "/usr/lib/python2.7/dist-packages/pyproj/__init__.py", line 358, in __new__
return _proj.Proj.__new__(self, projstring)
File "_proj.pyx", line 84, in _proj.Proj.__cinit__ (_proj.c:1170)
RuntimeError: no options found in 'init' file

seems there is nothing wrong with odm itself but between odm and python?

DChum17, 2018-05-24:
Hmm, well I removed JP2011 JP and it processed

[INFO]    Found 23 usable images
[DEBUG]   Line: EPSG:6672
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   Altitude data detected, enabling it for GPS alignment
[DEBUG]   ['use_exif_size: no', 'feature_process_size: 2048', 'feature_min_frames: 8000', 'processes: 4', 'matching_gps_neighbors: 8', 'optimize_camera_parameters: yes', 'use_altitude_tag: True', 'align_method: naive', 'bundle_use_gcp: yes']
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /var/www/data/5c53f56b-bcaf-4f03-9a93-111b7a1f7f15/opensfm

but then after running robust matching it gives this error

[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm create_tracks /var/www/data/5c53f56b-bcaf-4f03-9a93-111b7a1f7f15/opensfm
2018-05-23 23:33:48,916 INFO: reading features
2018-05-23 23:33:49,926 DEBUG: Merging features onto tracks
2018-05-23 23:33:50,645 DEBUG: Good tracks: 26305
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /var/www/data/5c53f56b-bcaf-4f03-9a93-111b7a1f7f15/opensfm
2018-05-23 23:33:52,722 INFO: Starting incremental reconstruction
Traceback (most recent call last):
File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
command.run(args)
File "/code/SuperBuild/src/opensfm/opensfm/commands/reconstruct.py", line 21, in run
report = reconstruction.incremental_reconstruction(data)
File "/code/SuperBuild/src/opensfm/opensfm/reconstruction.py", line 1115, in incremental_reconstruction
gcp = data.load_ground_control_points()
File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 525, in load_ground_control_points
fin, self.load_reference_lla(), exif)
File "/code/SuperBuild/src/opensfm/opensfm/io.py", line 423, in read_ground_control_points_list
projection = _parse_projection(lines.next())
File "/code/SuperBuild/src/opensfm/opensfm/io.py", line 408, in _parse_projection
raise ValueError("Un-supported geo system definition: {}".format(line))
ValueError: Un-supported geo system definition: EPSG:6672
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/run_opensfm.py", line 133, in process
(context.pyopencv_path, context.opensfm_path, tree.opensfm))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

I tried running again with EPSG:32653 which is the code for WGS84 53N but it throws the same error.

pierotofy, 2018-05-24:
I apologize, this is a bug. I've opened a related issue here: https://github.com/OpenDroneMap/OpenDroneMap/issues/828

In the meanwhile, the workaround is to either use a proj4 string definition, or use UTM (WGS84 UTM XX(N|S)).

Thank you for reporting the problem!

DChum17, 2018-05-29:
Thanks for the reply! Awaiting the update!

henryjiang88, 2018-06-13:
I've tried even with a proj4 string and it is giving the same error:

[DEBUG]   Line: ﻿+proj=utm +zone=17 +ellps=GRS80 +units=m +no_defs
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/dataset.py", line 86, in process
outputs.reconstruction = types.ODM_Reconstruction(photos, coords_file=tree.odm_georeferencing_gcp)
File "/code/opendm/types.py", line 140, in __init__
self.projection = self.parse_coordinate_system(coords_file)
File "/code/opendm/types.py", line 167, in parse_coordinate_system
return Proj(line.strip('\''))
File "/usr/lib/python2.7/dist-packages/pyproj/__init__.py", line 358, in __new__
return _proj.Proj.__new__(self, projstring)
File "_proj.pyx", line 77, in _proj.Proj.__cinit__ (_proj.c:1099)
File "_proj.pyx", line 370, in _proj._strencode (_proj.c:3939)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xef in position 0: ordinal not in range(128)

[gcp_list.txt](https://github.com/OpenDroneMap/WebODM/files/2098669/gcp_list.txt)

pierotofy, 2020-04-02:
GCPs have substantially improved since this issue was opened. Please retry with the latest versions of ODM/WebODM and re-open if the problem persists :clinking_glasses:

closed 2020-04-02 by pierotofy