WebODM/WebODM #457: Can't use WebODM

Greenlucas, 2018-05-23:
### How did you install WebODM?

Docker Boolbox

### What's your browser and operating system?

Firefox 60 on Windows 10
Chrome 66 on Windows 10

### What is the problem?

Can't open webodm in firefox or chrome. It looks fine in docker and everything starts normal and the only problem that pops up is ERROR:No containers to start

### What should be the expected behavior?

I should be able to access WebODM

### How can we reproduce this?

Just start WebODM

-----------------------------------------------------------------------------------------------------------
I might be missing something as i am new to all this stuff so please tell me if you need more information
-----------------------------------------------------------------------------------------------------------

Here is what i see in docker:

Komplett Gamer i35@Lucas-Dator MINGW64 /d/Program/Docker Toolbox/WebODM (master)
$ ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key:
SSL certificate:
SSL insecure port redirect: 80
Celery Broker: redis://broker
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting broker     ... done
Starting node-odm-1 ... done
Starting db         ... done
Starting worker     ... done
Starting webapp     ... done
ERROR: No containers to start
Creating network "webodm_default" with the default driver
Creating db         ... done
Creating broker     ... done
Creating node-odm-1 ... done
Creating worker     ... done
Creating webapp     ... done
Attaching to db, broker, node-odm-1, worker, webapp
db            | LOG:  database system was shut down at 2018-05-23 19:56:51 UTC
broker        | 1:C 23 May 19:57:02.203 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
db            | LOG:  MultiXact member wraparound protections are now enabled
broker        | 1:C 23 May 19:57:02.211 # Redis version=4.0.9, bits=64, commit=00000000, modified=0, pid=1, just started
db            | LOG:  database system is ready to accept connections
broker        | 1:C 23 May 19:57:02.211 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
broker        | 1:M 23 May 19:57:02.217 * Running mode=standalone, port=6379.
broker        | 1:M 23 May 19:57:02.220 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
worker        | Postgres is up - executing command
worker        | wait-for-it.sh: waiting 15 seconds for broker:6379
worker        | wait-for-it.sh: broker:6379 is available after 0 seconds
worker        | Checking for celery...   OK
webapp        | Postgres is up - executing command
webapp        | wait-for-it.sh: waiting 15 seconds for broker:6379
webapp        | wait-for-it.sh: broker:6379 is available after 0 seconds
webapp        |
webapp        |  _       __     __    ____  ____  __  ___
webapp        | | |     / /__  / /_  / __ \/ __ \/  |/  /
webapp        | | | /| / / _ \/ __ \/ / / / / / / /|_/ /
webapp        | | |/ |/ /  __/ /_/ / /_/ / /_/ / /  / /
webapp        | |__/|__/\___/_.___/\____/_____/_/  /_/
webapp        |
webapp        |
webapp        | Checking python version... 3.x, good!
webapp        | Checking GDAL version... GDAL 2.2.4, released 2018/03/19, excellent!
webapp        | Running migrations
node-odm-1    | info: No tasks dump found
node-odm-1    | info: Checking for orphaned directories to be removed...
worker        | Starting worker using broker at redis://broker
worker        | INFO Initializing GRASS engine using /usr/bin/grass74
webapp        | Generated secret key
webapp        | INFO Initializing GRASS engine using /usr/bin/grass74
webapp        | INFO Booting WebODM 0.5.2
webapp        | INFO Running npm install for posm-gcpi
webapp        | npm notice created a lockfile as package-lock.json. You should commit this file.
webapp        | npm WARN posm-gcpi@1.0.0 No description
webapp        | npm WARN posm-gcpi@1.0.0 No repository field.
webapp        |
webapp        | added 1 package in 6.473s
webapp        | INFO Registered [plugins.posm-gcpi.plugin]
webapp        | INFO Registered [plugins.measure.plugin]
webapp        | INFO Registered [plugins.osm-quickedit.plugin]
webapp        | Operations to perform:
webapp        |   Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
webapp        | Running migrations:
webapp        |   No migrations to apply.
webapp        | INFO Initializing GRASS engine using /usr/bin/grass74
webapp        | Checking for celery...   OK
webapp        | Scheduler has shutdown.
webapp        | Generating nginx configurations from templates...
webapp        | - nginx/nginx-ssl.conf
webapp        | - nginx/nginx.conf
webapp        | celery beat v4.1.0 (latentcall) is starting.
webapp        | [2018-05-23 19:57:16 +0000] [122] [INFO] Starting gunicorn 19.7.1
webapp        | [2018-05-23 19:57:16 +0000] [122] [INFO] Listening at: unix:/tmp/gunicorn.sock (122)
webapp        | [2018-05-23 19:57:16 +0000] [122] [INFO] Using worker: sync
webapp        | [2018-05-23 19:57:16 +0000] [132] [INFO] Booting worker with pid: 132
webapp        | INFO Initializing GRASS engine using /usr/bin/grass74
webapp        | __    -    ... __   -        _
webapp        | LocalTime -> 2018-05-23 19:57:16
webapp        | Configuration ->
webapp        |     . broker -> redis://broker:6379//
webapp        |     . loader -> celery.loaders.app.AppLoader
webapp        |     . scheduler -> celery.beat.PersistentScheduler
webapp        |     . db -> celerybeat-schedule
webapp        |     . logfile -> [stderr]@%WARNING
webapp        |     . maxinterval -> 5.00 minutes (300s)
webapp        |
webapp        |
webapp        | Congratulations! └@(･◡･)@┐
webapp        | ==========================
webapp        |
webapp        | If there are no errors, WebODM should be up and running!
webapp        |
webapp        | Open a web browser and navigate to http://localhost:8000
webapp        |

pierotofy, 2018-05-23:
Hey @Greenlucas what address are you using to open WebODM? Notice that it's not going to be `localhost`, you need to use the IP address you can find by running `docker-machine ip`.

Greenlucas, 2018-05-23:
i use the ip adress 192.168.99.100 that i got from running docker-machine ip