WebODM/WebODM #463: This project is not user－friendly

RuikangSun, 2018-06-11:
It takes me a long time to learn how to install docker. Is there any easier way to use WebODM?

smathermather, 2018-06-11:
Hi @RuikangSun -- docker abstracts away a lot of the complications of setting up WebODM. I recommend following this tutorial for getting started with docker: https://www.digitalocean.com/community/tutorials/how-to-install-and-use-docker-on-ubuntu-16-04

pierotofy, 2018-06-11:
Hi @RuikangSun,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-06-14 10:15) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2018-06-11:
Hi @RuikangSun, you can purchase an installer that automates the installation of docker and related dependencies from https://www.webodm.org if you are unable to install it manually.