WebODM/WebODM #468: gcp fail / UnicodeDecodeError: 'ascii' codec can't decode byte 0xef in position 0: ordinal not in range(128)

flohoff, 2018-06-18:
Docker installation of WebODM et al.

Used the GCP interface to tag some features and download the gcp file. Added task by selecting images and the gcp file. 

### What's your browser and operating system?

Firefox ESR Debian Stretch

### What is the problem?

Process fails parsing the gcp file.

[ ... ]
[DEBUG]   Loaded IMG_3413.JPG | camera: canon ixus 175 | dimensions: 5152 x 3864 | focal: 5.0 | ccd: None | lat: 52.3008305556 | lon: 8.59636666667 | alt: 225.0
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded IMG_3415.JPG | camera: canon ixus 175 | dimensions: 5152 x 3864 | focal: 5.0 | ccd: None | lat: 52.3011722222 | lon: 8.59774166667 | alt: 220.0
[INFO]    Found 112 usable images
[DEBUG]   Line: ﻿+proj=utm +zone=32 +ellps=WGS84 +datum=WGS84 +units=m +no_defs
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/dataset.py", line 86, in process
outputs.reconstruction = types.ODM_Reconstruction(photos, coords_file=tree.odm_georeferencing_gcp)
File "/code/opendm/types.py", line 140, in __init__
self.projection = self.parse_coordinate_system(coords_file)
File "/code/opendm/types.py", line 167, in parse_coordinate_system
return Proj(line.strip('\''))
File "/usr/lib/python2.7/dist-packages/pyproj/__init__.py", line 358, in __new__
return _proj.Proj.__new__(self, projstring)
File "_proj.pyx", line 77, in _proj.Proj.__cinit__ (_proj.c:1099)
File "_proj.pyx", line 370, in _proj._strencode (_proj.c:3939)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xef in position 0: ordinal not in range(128)

Flo

flohoff, 2018-06-18:
Thats the gcp file - As you can see there is a UTF-8 BOM at the beginning. 

<U+FEFF>+proj=utm +zone=32 +ellps=WGS84 +datum=WGS84 +units=m +no_defs  
472736.85       5794629.03      0       4428.00 1116.00 IMG_3338.JPG
472728.06       5794601.03      0       3573.08 752.45  IMG_3338.JPG
472808.95       5794631.87      0       4207.94 3130.99 IMG_3338.JPG
472819.22       5794645.50      0       4468.85 3382.71 IMG_3338.JPG
472824.11       5794562.00      0       2139.51 3304.35 IMG_3338.JPG
472484.19       5794586.65      0       3276.15 1265.81 IMG_3663.JPG
472484.19       5794586.65      0       2849.04 3240.91 IMG_3668.JPG
472462.33       5794611.81      0       4299.10 966.53  IMG_3663.JPG
472462.33       5794611.81      0       3920.00 2942.00 IMG_3668.JPG
472492.95       5794592.39      0       3327.62 1538.11 IMG_3663.JPG
472492.95       5794592.39      0       2923.15 3597.89 IMG_3668.JPG
472455.68       5794631.73      0       4901.32 1066.70 IMG_3663.JPG
472455.68       5794631.73      0       4568.00 3010.00 IMG_3668.JPG
472534.70       5794634.12      0       3942.52 3203.83 IMG_3663.JPG
472612.72       5794669.53      0       3888.30 1523.85 IMG_3653.JPG
472633.79       5794675.99      0       3905.32 2123.33 IMG_3653.JPG
472559.28       5794680.23      0       4439.85 438.71  IMG_3653.JPG
472601.03       5794705.36      0       4926.51 1580.13 IMG_3653.JPG
472633.89       5794713.73      0       4720.80 2493.64 IMG_3653.JPG
472649.80       5794668.37      0       3443.68 2450.76 IMG_3653.JPG
472413.81       5794602.04      0       4142.00 1540.00 IMG_3668.JPG
472388.94       5794595.84      0       4248.00 852.00  IMG_3668.JPG
472444.23       5794596.35      0       3556.00 2296.00 IMG_3668.JPG

Flo

pierotofy, 2018-06-18:
Hey @flohoff thanks for reporting the problem!

Is this limited to Firefox ESR or does it happen with other browsers also? Just trying to narrow it down.

bochen87, 2018-06-20:
@pierotofy I have the same issue - seems to be related to https://github.com/posm/posm-gcpi/issues/45 - it still generates 0 in the z values

pierotofy, 2018-06-20:
Mm, could you do a:

`./webodm.sh restart`

?

That will make sure you are using the latest version of the posm-gcpi plugin.

bochen87, 2018-06-21:
At the moment I'm using the windows installer - will switch to the linux version today and see if that problem still persists

bochen87, 2018-06-21:
@pierotofy I've installed the latest version on Linux and started up and run the restart command, it unfortunately still

pierotofy, 2018-06-21:
I think we might want to open an issue on https://github.com/posm/posm-gcpi/issues

The BOM character fix shoud be relatively easy: https://github.com/eligrey/FileSaver.js/issues/160

The Z value I'm not entirely sure, should the GCP interface output the Z column even though it's always 0? @smathermather

smathermather, 2018-06-21:
If not otherwise specified, it should output 0. With some work, one could extract better values from SRTM or other near global DEM source, but with mapzen shut down, I don’t know offhand what service that would be unless some clever combo of curl, GDAL, and NASA’s SRTM archive.

bochen87, 2018-06-22:
Are there any other ways to produce a proper GCP file?

pierotofy, 2018-06-22:
You can manually select the text from the GCP interface and do CTRL+C instead of using the "Save" button. That should leave the BOM character out.

pierotofy, 2018-06-28:
This is now fixed.

closed 2018-06-28 by pierotofy