WebODM/WebODM #469: Add Instructions to run WebODM at startup

pierotofy, 2018-06-18:
Could be useful for some users that want to run it in the cloud.

ChipwizBen, 2019-02-27:
My little contribution for systemd based hosts:

`cat << _EOF_ > /etc/systemd/system/WebODM.service 
[Unit]
Description=Web Open Drone Map
After=docker.service

[Service]
Type=oneshot
ExecStartPre=usr/bin/echo never > /sys/kernel/mm/transparent_hugepage/enabled
ExecStart=/opt/WebODM/webodm.sh start --cleanup_tasks_after 1440
ExecStop=/opt/WebODM/webodm.sh stop
RemainAfterExit=yes

[Install]
WantedBy=default.target
_EOF_

systemctl daemon-reload
systemctl enable WebODM
`

pierotofy, 2021-02-03:
Docker by default launches containers at startup.

closed 2021-02-03 by pierotofy

implan-sig, 2022-05-20:
ChipwizBen solution worked great for me!
I just had to remove the /usr/bin path and run echo directly.
I am running WebODM over Google Cloud Platform and Ubuntu 22.04

chemlabros, 2025-01-26:
If you installed WebODM on linux under a docker image (tested in Ubuntu 20.04) in order to start this docker container on boot you need to be sure that: 

1. Docker daemon starts at boot. If this doesn't happen add a file at /etc/systemd/system/mydocker.service with content:
```
[Unit]
Description=Docker Daemon
After=network.target

[Service]
ExecStart=/opt/docker.sh

[Install]
WantedBy=multi-user.target
```

This is calling file /opt/docker.sh on boot starting docker.

2. In /opt folder put a file docker.sh with content (this is called from the mydocker.service file on boot):
`sudo service docker start `

3. In /opt folder put the WebODM folder and change it's ownership to root:root in a recursive manner (folder and files):
sudo chown -R root:root /opt/WebODM

4. Add a file at /etc/systemd/system/WebODM.service with the content as provided from @ChipwizBen above. 
This starts WebODM service on boot

5. After you change any files - adding services - to /etc/systemd/system folder use the command below for the changes to apply: 
`sudo systemctl daemon-reload`

This is the equivalent of starting ubuntu, starting docker from command line and giving the webodm.sh start command. If docker isn't present when you call webdom.sh of course webodm won't start. 
You can run it nativelly but it isn't easy when installing and when updating. Been there just so you know.

Really like this software so in the future I will be buying the standalone .exe (although I don't need it) in order to support the software.

Please mark this issue as closed.