WebODM/WebODM #470: Webodm natively install

hildermesmedeiros, 2018-06-18:
### How did you install WebODM? (Docker, natively, ...)?

natively

### What's your browser and operating system?

Chrome 65 on Linux

### What is the problem?

Cant log in Webodm

### What should be the expected behavior?

**Exception Value: | [Errno 2] No such file or directory:** '/home/odm/WebODM/app/media/settings/logo512_aWw2zNw.png'
-- | --
/home/odm/env/lib/python3.5/site-packages/django/core/files/storage.py in _open, line 218

Request Method: | GET
-- | --
http://localhost:8000/login/
2.0.3
FileNotFoundError
[Errno 2] No such file or directory: '/home/odm/WebODM/app/media/settings/logo512_aWw2zNw.png'
/home/odm/env/lib/python3.5/site-packages/django/core/files/storage.py in _open, line 218
/home/odm/env/bin/python
3.5.2
['/home/odm/WebODM',  '/home/odm/env/lib/python35.zip',  '/home/odm/env/lib/python3.5',  '/home/odm/env/lib/python3.5/plat-x86_64-linux-gnu',  '/home/odm/env/lib/python3.5/lib-dynload',  '/usr/lib/python3.5',  '/usr/lib/python3.5/plat-x86_64-linux-gnu',  '/home/odm/env/lib/python3.5/site-packages',  '/home/odm/WebODM']
Seg, 18 Jun 2018 15:18:38 -0300

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I do not know. I supposse something went wrong and can't figure out what. I suspect somethin is faling or I forgot to configure some file.

hildermesmedeiros, 2018-06-18:
ps: 
I arrived at the login page, although it took me 5 minutes to enter.

I've found the problem (I think....), it takes some time to build the the page.
and for some reason i could try to log in before the page was ready...

pierotofy, 2020-04-02:
If this is still a problem, let's move the conversation over to https://community.opendronemap.org :+1:

closed 2020-04-02 by pierotofy