WebODM/WebODM #472: Processing Error

dodge-ttu, 2018-06-22:
Hello,

I've run ODM through docker many times without issue but recently ran into a problem on two consecutive attempts to generate an orthomap from about 800 images. Large composites have worked fine before but the last two attempts have been problematic.

I'm running ODM through docker on Ubuntu installed on a very powerful desktop computer. I receive the following error:

[DEBUG]   running /code/SuperBuild/install/bin/texrecon /code/opensfm/reconstruction.nvm /code/odm_meshing/odm_mesh.ply /code/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/code/SuperBuild/install/bin/texrecon (built on Jun  4 2018, 13:59:47)
Load and prepare mesh: 
PLY Loader: comment PCL generated
Reading PLY: 100058 verts... 199898 faces... done.
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 1288
NVM: Number of features: 0
        Loading 0%...terminate called after throwing an instance of 'util::Exception'
  what():  /code/opensfm/undistorted/batch2: No such file or directory
Aborted (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 47, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 139, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134

Perhaps someone can help me decipher this error message and point me in the right direction.

My call to execute ODM from the command line is as follows:

sudo docker run -it --rm -v "/home/will/Temp Drone Image Folder/2018-06-21_75_75_20_rainMatrix:/code/images" -v "/home/will/Temp Drone Image Folder/2018-06-21_75_75_20_rainMatrix/odm_orthophoto:/code/odm_orthophoto" opendronemap/opendronemap --orthophoto-resolution 120 --force-ccd 13.2

pierotofy, 2018-06-22:
Hi @dodge-ttu,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-06-25 09:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

dodge-ttu, 2018-06-23:
The problem here, as I finally discovered, is that I my image filenames had characters that caused the process to fail. 

The filenames were batch2 (2).jpg, batch (3).jpg, batch (4).jpg, etc. This is a result of taking over 1000 images with a DJI drone where the naming starts over at 1000 and images are stored in separate folders during flight. I was forced to batch change the filenames so that they could all be in the same folder. Windows always adds a parenthesis and a space which caused the failure. When I used these filenames in Ubuntu the process failed.

pierotofy, 2018-06-23:
Thanks for reporting the problem @dodge-ttu! This is something that we should fix (or at a minimum warn in a more user-friendly manner).

garlac, 2018-07-30:
Thank you @dodge-ttu . I faced the same issue. Started processing again and hope i will get proper output.

Could you please help me in knowing how you figured it out? ..so that I can sort out any such issues in future. thanks!

dodge-ttu, 2018-07-30:
@garlac I took a ten image sample set and began trial and error. The file names stood out becuase they contained some potential problematic characters. 

My trouble shooting at this point is nothing more than trial and error with small test sets.

pierotofy, 2020-04-02:
This is now fixed with the latest version of ODM (filenames with spaces are handled). Please update and try again (or re-open if the problem persist). :clinking_glasses:

closed 2020-04-02 by pierotofy