WebODM/WebODM #479: [Feature Request] Exporting Projects to load in another WebODM

kornyhiv, 2018-06-25:
Using WebODM on multiple cloud servers for fast processing. Any way to include (in the future) a way to export projects to then import them on a different WebODM installation?

I do not want to use Nodes, as these systems are independent. I would like to be able to export completed projects and then add/import them to the WebODM installed on my laptop for usage/utilization.

Thank you.

pierotofy, 2018-06-25:
Hi @kornyhiv,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-06-28 17:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2018-06-26:
This is a great feature request. +1.

rdweaver, 2018-10-19:
> In my case I imagine running a job on a powerful server at home and then importing them to a less powerful public web server to host the output.
@bertramt That is exactly the workflow I would like to do as well. Would it be possible to have a processing node on your local server and connect your public server to that? Does the processing node have to be online to view the assets etc, after it is processed?

bertramt, 2018-10-19:
@rdweaver Fairly sure you could connect to a a public server to a home processing node but you would either need a vpn between the machines or to expose ports on your home network to the internet.  Both things in my case I'd prefer to avoid.

kornyhiv, 2018-10-19:
I believe the easiest way to accomplish this, would be to gzip the entire project files, and then import than into another WEBODM installation, and have the server restore the data project with the database.
Obviously creating some type of script to do that, and editing the website to accommodate an upload project option, etc.
This does not sound too difficult to accomplish. I wish I could help.

pierotofy, 2019-02-21:
People can now import existing tasks that have been exported via "Downloads Assets" -- "All Assets".

closed 2019-02-21 by pierotofy

smathermather, 2019-02-22:
Spectacular.