WebODM/WebODM #482: Erroneous EXIF errors and warnings

DaveSprague, 2018-06-30:
### How did you install WebODM? (Docker, natively, ...)?

I did a manual install of WebODM  according to the instructions provided.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 67 on Windows 10 Pro version 1803, build 17134.112

### What is the problem?

When I run a task that includes resizing the source images, I get a very long list of EXIF related errors and warnings in the log.  Here is an example:

`Error: Directory (Last IFD item) with 19712 entries considered invalid; not read.
Error: Directory (Last IFD item): Next pointer is out of bounds; ignored.
Warning: Directory (Last IFD item), entry 0x0000 has unknown Exif (TIFF) type 1114; setting type size 1.
Error: Directory (Last IFD item), entry 0x0000 has invalid size 1145260365*1; skipping entry.
Warning: Directory (Last IFD item), entry 0x4d45 has unknown Exif (TIFF) type 17481; setting type size 1.
Error: Directory (Last IFD item), entry 0x4d45 has invalid size 1096565834*1; skipping entry.
Warning: Directory (Last IFD item), entry 0x3431 has unknown Exif (TIFF) type 11850; setting type size 1.
Error: Directory (Last IFD item), entry 0x3431 has invalid size 1346830406*1; skipping entry.
Warning: Directory (Last IFD item), entry 0x5300 has unknown Exif (TIFF) type 0; setting type size 1.
Error: Offset of directory (Last IFD item), entry 0x5300 is out of bounds: Offset = 0x00010000; truncating the entry
`

### What should be the expected behavior?

According to @pierotofy, these errors "are due to WebODM using a different library for resizing the images than what OpenDroneMap uses,  The error is more of a “warning” and has no effect on the output dataset (the EXIF entries that we care about, lat/lon/altitude/focal length/sensor size are all read properly)."

Based on that, I believe that the expected behavior would be to remove or suppress these error and warming messages.

If some one can tell me what library, used by webODM, is generating these messages I'd be happy to look at ways to correct this.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I believe these errors occur when using webODM and selecting an the option to resize the images.

pierotofy, 2018-10-25:
This should now be fixed.

pierotofy, 2020-12-16:
This has been fixed in ODM some time ago! :clinking_glasses:

closed 2020-12-16 by pierotofy