WebODM/WebODM #485: [Question / Improvement] [GCPi] Possibility of Adding Points Manualy Instead of Using the Map.

PR91, 2018-07-02:
Hello All,

I am currently using WebODM to do some mapping with ground control points. Currently, as far as I am aware, we need to use the map to select our ground control points that match the picture, the problem is that the satelite view is not precise enough to select GCP. I did some research and the tool that is currently available (GCPi), the user used to be able to add GCP by inputing the coordinates. Is that functionality gone? 

That would help me alot.

Thank you for your time.

pierotofy, 2018-07-02:
Hi @PR91,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-07-05 06:10) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

PR91, 2018-07-02:
I will follow your sugestion, sorry for the incovinience. Ive seen some sugestions in here, so I thought It was OK.

closed 2018-07-02 by PR91

smathermather, 2018-07-02:
Ah, this is a request for our under-documented feature. I can’t respond at the moment, but could you bug me in a week if I haven’t followed up? I need to write documentation on how to input coordinates for GCPs.

smathermather, 2018-07-02:
Also, best to post this question on https://community.opendronemap.org. Thanks!