WebODM/WebODM #490: ERROR: unauthorized: authentication required

fvila5, 2018-07-16:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 67 on Windows 10

### What is the problem?
When traying to install webodm:

$ ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
SSL: NO
SSL key:
SSL certificate:
SSL insecure port redirect: 80
Celery Broker: redis://broker
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Starting broker     ... done
Starting node-odm-1 ... done
Starting db         ... done
Starting worker     ... done
Starting webapp     ... done
ERROR: No containers to start
Pulling worker (opendronemap/webodm_webapp:latest)...
latest: Pulling from opendronemap/webodm_webapp
cc1a78bfd46b: Pull complete
6861473222a6: Pull complete
7e0b9c3b5ae0: Pull complete
3ec98735f56f: Pull complete
9b311b87a021: Downloading [====================>                              ]  88.45MB/213.2MB
048165938570: Download complete
1ca3d78efb22: Download complete
0f6c8999c3b7: Download complete
5a85410f5000: Download complete
d663fec369c0: Download complete
b5ed80220591: Download complete
df4fa42da4dd: Downloading
cf8acae8f635: Downloading
57b308906a0f: Downloading
cb573624f515: Downloading
4f85140a8b99: Downloading
4780432831e0: Downloading
774ae97c3547: Downloading
f3a02973b80f: Waiting
47775935e591: Downloading
5310fcd17ffd: Waiting
58f6b75d9dc8: Waiting
4b344285be11: Waiting
4dc8e330d406: Waiting
749edf13f121: Waiting
eeb68df61844: Waiting
**ERROR: unauthorized: authentication required**

What should I do? Please

fvila5, 2018-07-16:
Solved with login docker. Closed

closed 2018-07-16 by fvila5

alfadona90, 2018-12-26:
@fvila5 hai, i face same problem with you.. can you tellme how to solve that with docker login? i dont understand.. 

im newbie

fvila5, 2018-12-27:
Hello,

in the terminal put **$ docker login**
and will ask you your username and password of Docker Hub.

alfadona90, 2018-12-27:
i was login for many time n logut then login again, still have same problem  with ERROR: unauthorized: authentication required..