WebODM/WebODM #494: Document Plugin API

pierotofy, 2018-07-30:
As the plugins API starts to stabilize, we need to start documenting it so that people can utilize it. At a minimum links to the example plugins should be provided.

pierotofy, 2018-08-11:
Added a brief description in the README.

epurnell, 2018-12-23:
Would be great to promote this to a higher priority as I would like to write a few plugins for WebODM.

pierotofy, 2018-12-23:
That's awesome @epurnell ! In the meanwhile, have you tried to look at the examples? https://github.com/OpenDroneMap/WebODM/tree/master/plugins

pierotofy, 2019-01-07:
+1 for this.

smathermather, 2019-01-07:
@petrasovaa has requested this as well via email.

pvgmenegasso, 2021-05-21:
Hi, I'm also interested on adding plugins to the application, and would greatly appreciate any help with documentation. I've already looked at the source codes and all the links, both on readme and on the documentation.