WebODM/WebODM #497: Process images without GPS Altitude date

Ivanhoe81, 2018-08-10:
The current version cannot process georeferenced orthophoto from images without valid GPS Altitude value in the Exif tag. 

I'm running a server and providing image processing service to my clients for a year. Yesterday I updated WebODM and I'm facing this issue which was not a problem in the older version.

I modified the 'odm_extract_utm' binary to avoid the error messages and to parse the latitude and longitude data and add 0 value for altitude (as it was in the older version) and to put them into coords.txt. It works fine but there's a problem with the georeferencing: the orthophoto is mirrored (on Z axis?).

What can be done to work as before?

pierotofy, 2018-08-10:
Hi @Ivanhoe81,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-08-13 07:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Ivanhoe81, 2018-08-10:
**How did you install WebODM? (Docker, natively, ...)?**
Docker on Ubuntu 16.04

**What's your browser and operating system?** (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome 68 on Windows 7
Browser independent issue.

**What is the problem?**
The current version cannot process georeferenced orthophoto from images without valid GPS Altitude value in the Exif tag.

I'm running a server and providing image processing service to my clients for a year. Yesterday I updated WebODM and I'm facing this issue which was not a problem in the older version.

I modified the 'odm_extract_utm' binary to avoid the error messages and to parse the latitude and longitude data and add 0 value for altitude (as it was in the older version) and to put them into coords.txt. It works fine but there's a problem with the georeferencing: the orthophoto is mirrored (on Z axis?).

**What should be the expected behavior?**
Process images without GPS Altitude value and create georeferenced orthophoto.

**How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)**
Process the images on the link below:
https://app.box.com/s/hcjbayd8e1m8nlo7wb8nncds38nj18eh

Processing options: 
Options: min-num-features: 15000, orthophoto-no-tiled: true, build-overviews: true, verbose: true, use-exif: true, use-pmvs: true

Modified odm_extract_utm:
No error on missing GPS altitude, 0 value for altitude in coords.txt
https://app.box.com/s/kiv89gqaquoimrvep4gqj4uxq5egw4p3

pierotofy, 2020-04-02:
Confirm that latest versions of ODM/WebODM can process this (even without proper georeferencing/altitude):

[image]

closed 2020-04-02 by pierotofy