WebODM/WebODM #498: Straight lines turn into curved lines in orthomosaic

martinsoko, 2018-08-10:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of Your web browser is:
Edge 42 on Windows 10 (64-bit)
Is your Web browser up to date?
✓ Your web browser is up to date.)

### What is the problem?

The orthomosaic is very distorted. Straight lines appear as curved surfaces.

How it looks using WebODM:

[image]

How it should look: (result obtained using Pix4D)

[image]

I've tried with different settings but I couldn't get rid of that weird curves.
The settings I've tried include Fast orthophoto, High resolution and custom settings such as:

    {'name': "resize-to", 'value': -1},
    {'name': "orthophoto-resolution", 'value': 150},
    {'name': "fast-ortophoto", 'value': "false"},
    {'name': "min-num-features", 'value': 12000},
    {'name': "dem-terrain-type", 'value': "ComplexNonForest"}

I get better or worse resolution but in all cases I couldn't manage to get a result similar to the one I got using Pix4D.

### What should be the expected behavior?

I should get a nicer orthomosaic with straight lines.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

pierotofy, 2018-08-11:
Hey @martinsoko :hand:, thanks for reporting the problem. Any chance you could share your images with us so that we can try to reproduce the problem? It's the first time I see this particular type of distortion.

martinsoko, 2018-08-13:
Hi @pierotofy, I just uploaded the dataset to GoogleDrive. I edited the original comment to include it.

https://drive.google.com/open?id=12eqSUndyyhgSfrprnUzNC-3YuBsnsGZB

Thanks in advance!

pierotofy, 2018-08-13:
The distortion is caused by "holes" or bumps in the reconstructed mesh.

[image]

I was able to get a better result using the upcoming changes in ODM that were integrated as part of https://github.com/OpenDroneMap/OpenDroneMap/projects/2 (which haven't been released yet), but it's still not perfect.

[image]

With the current version of ODM, I could recommend trying to enable the `--fast-orthophoto` flag, or fly at higher elevation (not sure if this is an option, but I think it would improve results). Pix4D's method of texture blending is different than ours and seem to work very well for this particular dataset. Also with the upcoming changes to ODM you should be able to soon see some improvements.

Future work on improving our texturing approach could also help improve the quality of this particular dataset.

martinsoko, 2018-08-13:
OK then, I'll wait for the next release. I've already tried running the task with the flag `--fast-orthophoto` but the results were the same.

Do you have an estimated release date?

Thanks a lot for your time and dedication!!

pierotofy, 2018-08-13:
No official release date yet, but hopefully very soon. Keep an eye on our blog for announcements at https://www.opendronemap.org :+1:

pierotofy, 2020-04-02:
This should have improved substantially with the latest changes to ODM. :clinking_glasses:

closed 2020-04-02 by pierotofy