WebODM/WebODM #502: Feature Request: Google Drive access

Vantage-DS, 2018-08-17:
To give an option to 'Select Images' from Google Drive when submitting images.

As post - http://community.opendronemap.org/t/importing-from-google-drive/924

I'm not sure if this helps in any way at all, but there's an add-on to owncloud which may have code you can use? - https://github.com/owncloud/files_external_gdrive
Sorry If that's absolutely useless, I'm not a developer :P

pierotofy, 2018-08-17:
Hi @Vantage-DS,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-08-20 09:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2024-09-28:
No bandwidth for this; we'd welcome a pull request. :pray:

closed 2024-09-28 by pierotofy