WebODM/WebODM #509: Add the ability to mount WebODM on a prefix to co-exist with other services

mojodna, 2018-08-31:
Allowing WebODM to sit on `/webodm/` will allow it to coexist with other services without needing to configure DNS, etc.

This partially works by including this in `local_settings.py`:

```python
# triggers Django to include /webodm in generated paths
FORCE_SCRIPT_NAME = 'webodm'
STATIC_URL = '/webodm/static/'
```

and by configuring the outermost Nginx with:

```
  location /webodm {
    # notice the trailing slash; this prevents Nginx from including /webodm when making requests
    proxy_pass http://127.0.0.1:11111/;
  }

  location /media {
    proxy_pass http://127.0.0.1:11111;
  }

  location /plugins {
    proxy_pass http://127.0.0.1:11111;
  }
```

It would be great to not have `media` and `plugins` polluting the root namespace; https://github.com/OpenDroneMap/WebODM/blob/0fa17a79661993af8e40ce3acc9120b43d5def6e/app/templatetags/plugins.py#L7-L17 is one place that needs to become prefix-aware.

I was unable to get any further, as the login page attempts to `POST` to a relative path (which resolves to `/webodm/login/webodm/login`) courtesy of https://github.com/OpenDroneMap/WebODM/blob/0fa17a79661993af8e40ce3acc9120b43d5def6e/app/templates/app/registration/login.html#L10

pierotofy, 2018-08-31:
Hi @mojodna,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-09-03 19:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2018-09-01:
Hey @mojodna :hand:!

We would totally welcome a PR for supporting prefixed URLs.

A few tricky areas that will likely need attention, along with the areas you've already identified:
 - Some CSS absolute URLs
 - Some Javascript modules use absolute URLs for REST requests.

On a side note, have you thought about running the two applications on separate ports instead?