WebODM/WebODM #515: JavaScript exceptions on view map: "TypeError: Cannot create property '_leaflet_id' on string 'Google Maps Hybrid'"

DarylWM, 2018-09-06:
### How did you install WebODM? (Docker, natively, ...)?

Docker - macos installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

https://whatismybrowser.com/w/BGTEXBP

### What is the problem?

Since upgrading to the latest release using WebODM Manager 1.1, all my maps are completely blank when I 'view map'. 
[image]

These exceptions appear in the JS console:
[image]

### What should be the expected behavior?

I should see my map.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I'm using the default processing settings. This was working fine a week ago.

pierotofy, 2018-09-06:
Seems like a problem introduced in #514. A workaround for now is to use Google Chrome instead of Safari. We'll have a fix for this by today.

pierotofy, 2018-09-06:
This should be now fixed.

closed 2018-09-06 by pierotofy

DarylWM, 2018-09-06:
Yes that has resolved the issue. I really appreciate the fast fix!