WebODM/WebODM #519: Add Information / Diagnostic page with version info

pierotofy, 2018-09-13:
This way users can find out which version of WebODM they are running.

xmutantson, 2018-09-13:
Is there any way to find out the version as it is? I updated my processing node and then ran a job i ran before with the default settings and it took the same amount of time and was still using pvms. I deleted everything and did a fresh installation and it still appears that I don't have the new version. I'm using WebODM manager for windows.

pierotofy, 2018-09-13:
Pressing the "Update" button from the Manager should take care of updating everything. You should be able to confirm that you have the latest version if you press the "Open Shell" button from the Maintenance panel and type:

```
docker run --rm --entrypoint cat opendronemap/node-opendronemap /code/VERSION
```

If you get `0.4` then you have the latest version.

xmutantson, 2018-09-14:
The output was 0.4.0, but when I try to customize settings in webodm, the pvms settings are still there, the job took just as long as the old version I tested, the option of texturing-nadir-weight was not present, orthophoto resolution is still pixels/meter, the majority of process time seems to be ceres solving, and overall it looks as though no changes had an effect, even though I deleted everything and started over. Maybe I should have brought this up as its' own issue but I found this while trying to see if the manager was pulling the wrong version.

pierotofy, 2018-09-14:
Something is not right. From the shell, type:

```
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml down
```

Then restart WebODM Manager. You should see the `texturing-nadir-weight` option.

xmutantson, 2018-09-14:
That option should be under the customization menu of a task in WebODM, correct? I don't see that option.

pierotofy, 2018-09-14:
Are there any errors in the console tab of Manager when you press Update? Could you copy paste the output here?

xmutantson, 2018-09-14:
C:\WebODM Manager>cd WebODM/ 
C:\WebODM Manager\WebODM>git pull origin master 
From https://github.com/OpenDroneMap/WebODM
 * branch            master     -> FETCH_HEAD
Already up-to-date.
C:\WebODM Manager\WebODM>docker-compose down --remove-orphans 
Stopping webapp     ... 
Stopping worker     ... 
Stopping db         ... 
Stopping broker     ... 
Stopping node-odm-1 ... 
Stopping webapp ... 
Stopping worker ... 
Stopping db     ... 
Stopping broker ... 
[31mwebapp        |[0m INFO Initializing GRASS engine using /usr/bin/grass74
[4A[2K
Stopping webapp ... [32mdone[0m
[5A[2K
Stopping webapp     ... [32mdone[0m
[31mwebapp exited with code 137
[0m[36mnode-odm-1    |[0m info: Server has started on port 3000
[36mnode-odm-1    |[0m info: Closing server
[36mnode-odm-1    |[0m info: Exiting...
[5B[1A[2K
Stopping node-odm-1 ... [32mdone[0m
[36mnode-odm-1 exited with code 0
[0m[35mworker        |[0m INFO Cleaning up OAM datastore for task 303cb68d-73a4-4eff-bcc2-fe25378384fa
[35mworker        |[0m [2018-09-14 03:11:26,543: INFO/ForkPoolWorker-2] Cleaning up OAM datastore for task 303cb68d-73a4-4eff-bcc2-fe25378384fa
[1B[4A[2K
Stopping worker     ... [32mdone[0m
[4B[3A[2K
Stopping worker ... [32mdone[0m
[35mworker exited with code 137
[0m[32mbroker        |[0m 1:M 14 Sep 03:46:45.196 * Background saving terminated with success
[32mbroker        |[0m 1:signal-handler (1536896869) Received SIGTERM scheduling shutdown...
[33mdb            |[0m LOG:  autovacuum launcher started
[32mbroker        |[0m 1:signal-handler (1536896869) Received SIGTERM scheduling shutdown...
[32mbroker        |[0m 1:M 14 Sep 03:47:49.257 # User requested shutdown...
[32mbroker        |[0m 1:M 14 Sep 03:47:49.257 * Saving the final RDB snapshot before exiting.
[33mdb            |[0m LOG:  received smart shutdown request
[33mdb            |[0m LOG:  autovacuum launcher shutting down
[33mdb            |[0m LOG:  shutting down
[32mbroker        |[0m 1:M 14 Sep 03:47:49.471 * DB saved on disk
[32mbroker        |[0m 1:M 14 Sep 03:47:49.471 # Redis is now ready to exit, bye bye...
[33mdb            |[0m LOG:  database system is shut down
[32mbroker exited with code 0
[3B[1A[2K
Stopping broker ... [32mdone[0m
[4B[2A[2K
Stopping broker     ... [32mdone[0m
[1B[2A[2K
Stopping db     ... [32mdone[0m
[2B[3A[2K
Stopping db         ... [32mdone[0m
[3B
[5B
[2BRemoving webapp ... 
Removing worker ... 
Removing db     ... 
Removing broker ... 
[1A[2K
Removing broker ... [32mdone[0m
[1B[3A[2K
Removing worker ... [32mdone[0m
[3B[4A[2K
Removing webapp ... [32mdone[0m
[4B[2A[2K
Removing db     ... [32mdone[0m
[2BRemoving network webodm_default
[0m[33mdb exited with code 0
[0m
C:\WebODM Manager\WebODM>exit 0 
C:\WebODM Manager\WebODM>docker pull opendronemap/node-opendronemap 
Using default tag: latest
WebODM has started. You can close this window now. WebODM will continue running.
Error response from daemon: Get https://registry-1.docker.io/v2/opendronemap/node-opendronemap/manifests/latest: dial tcp: lookup registry-1.docker.io on 10.0.2.3:53: read udp 10.0.2.15:56380->10.0.2.3:53: i/o timeout
C:\WebODM Manager\WebODM>docker pull opendronemap/webodm_db 
Using default tag: latest
latest: Pulling from opendronemap/webodm_db
Digest: sha256:437e8772fddeb7e740fe38ca8213eaee6b05584104efc6c75eb26f9f7da1da9d
Status: Image is up to date for opendronemap/webodm_db:latest
C:\WebODM Manager\WebODM>docker pull opendronemap/webodm_webapp 
Using default tag: latest
latest: Pulling from opendronemap/webodm_webapp
Digest: sha256:c465e1128e84fb5215d189ce456c99f31d9dce04331a95dded90837da825fa7d
Status: Image is up to date for opendronemap/webodm_webapp:latest
C:\WebODM Manager\WebODM>exit 0 
Update successful :) Press "Start WebODM" to continue.

pierotofy, 2018-09-14:
Thank you, this helps a lot. This looks like a problem with the Manager code. I will have an update released soon to fix it. In the meanwhile, the workaround is to issue `docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml down` from the shell (press the "Open Shell" button from the maintenance panel and type that line). Then restart WebODM Manager.

pierotofy, 2018-11-02:
This page should also show node-odm/odm version info (or add that information somewhere else, perhaps in the processing node info page).

pierotofy, 2018-11-03:
Done. Will take care of  node-odm / odm version info separately. https://github.com/OpenDroneMap/node-OpenDroneMap/issues/50

closed 2018-11-03 by pierotofy