WebODM/WebODM #522: [Feature] Duplicate a task / image set

danbjoseph, 2018-09-20:
I would like to be able to duplicate a task so that WebODM copies the files from one folder on the server to another, and I don't have to re-upload. My initial use-case is that I'm reprocessing a dataset with different parameters and want to preserve the original outputs, log, time to process, etc. within WebODM.

(_see [Copy images to new project without reuploading?](http://community.opendronemap.org/t/copy-images-to-new-project-without-reuploading/1094) on the forum_)

pierotofy, 2018-09-20:
Hi @danbjoseph,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-09-22 21:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

dawonn-haval, 2019-10-04:
me2.