WebODM/WebODM #523: How to prepare photos with location information for webdom

david69cn, 2018-09-20:
My UAV is loaded with RTK equipment. The exposure time of the photo and the location of the GPS are synchronized. However, the location information of GPS is recorded in text file alone. A lot of photos, how to write photos in batches of GPS latitude and longitude information

pierotofy, 2018-09-20:
Hi @david69cn,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-09-23 05:15) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2018-09-20:
I saw the post on community, which is the appropriate place for this question, so I’m going to close this issue. Thanks!

closed 2018-09-20 by pierotofy