WebODM/WebODM #526: Unclear error in Firefox Quantum 62.0 on Windows 10

smathermather, 2018-09-25:
### How did you install WebODM? (Docker, natively, ...)?

./webodm.sh start

### What's your browser and operating system?
Firefox 62 on Windows 10

### What is the problem?

After images are selected in "Select Images and GCP" dialog, if I select invalid files in addition to valid ones, I receive the error "Could not upload all files. An error occurred. Please try again." shows in an orange alert at the top of the dialog. But, if one chooses options and clicks "Start Processing" the image uploads will begin.

### What should be the expected behavior?

A clearer message such as "Could not upload all files. Some files are not image files (jpg or png) and will not be used" would be helpful. This does not happen in Chrome, perhaps due to chrome pre-filtering the image list?

### How can we reproduce this?

1. Use Firefox or Brave
2. Select both image and non-image files
3. See error.

pierotofy, 2018-09-25:
Thanks @smathermather! Like you say, this needs a better error message.

What I'm wondering, is this error specific only to Firefox/Brave? Does Chrome just ignore the invalid files?

smathermather, 2018-09-25:
It doesn’t happen in chrome.

pierotofy, 2018-09-28:
This is now fixed

closed 2018-09-28 by pierotofy

smathermather, 2018-09-28:
w00t!