WebODM/WebODM #531: Unreferenced models processing in WebODM but not displaying

smathermather, 2018-10-01:
### How did you install WebODM? (Docker, natively, ...)?

Docker: ./webodm.sh start

### What's your browser and operating system?

Chrome 69 on Windows 10

### What is the problem?

Unreferenced datasets appear to be processed, but not displayed in "View 3D Model".

### What should be the expected behavior?

I would anticipate that projects that can successfully create a point cloud can be viewed.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

1. Download [this dataset](https://clemetparks-my.sharepoint.com/:f:/g/personal/svm_clevelandmetroparks_com/EhEvKIm_h2lLiSXil96GaYkBrP04qGTHwSoR2Zca6CtXUQ?e=6WWytb) 
1. Upload into WebODM
1. Drink some coffee or beverage of choice.
1. When complete, *View 3D Model*
1. Receive error: ```Could not load point cloud. This task doesn't seem to have one. Try processing the task again.```

smathermather, 2018-10-01:
Here's its appearance when processed through ODM and viewed in Meshlab:
[image]

pierotofy, 2018-10-03:
Thanks @smathermather, do you have a copy of the task output by any chance?

smathermather, 2018-10-03:
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/images
[DEBUG]   /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/images
[DEBUG]   Loaded IMG_9938.JPG | camera: canon powershot s100 | dimensions: 2048 x 1536 | focal: 5.2 | ccd: 5.33 | lat: None | lon: None | alt: None

...

[INFO]    Running ODM SMVS Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_mesh.ply
[DEBUG]   running /code/SuperBuild/src/PoissonRecon/Bin/Linux/PoissonRecon --in /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/smvs/smvs_dense_point_cloud.ply --out /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_mesh.dirty.ply --depth 9 --pointWeight 4 --samplesPerNode 1.0 --threads 16 --linearFit
[DEBUG]   running /code/build/bin/odm_cleanmesh -inputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_mesh.dirty.ply -outputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_mesh.ply -removeIslands -decimateMesh 100000
[DEBUG]   Writing ODM 2.5D Mesh file in: /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_25dmesh.ply
[WARNING] Cannot calculate GSD, using requested resolution of 5.0
[INFO]    Created temporary directory: /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp
[INFO]    Creating DSM for 2.5D mesh
[INFO]    Creating ../var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp/mesh_dsm_r0.141421356237 [idw] from 1 files
[DEBUG]   running pdal pipeline -i /tmp/tmpAuaTdU.json > /dev/null 2>&1
[INFO]    Completed ../var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp/mesh_dsm_r0.141421356237 [idw] in 0:00:12.575316
[INFO]    Starting gap-filling with nearest interpolation...
[INFO]    Completed gap-filling to create ../var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp/mesh_dsm.tif in 0:04:06.956609
[INFO]    Sampling points from DSM: /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp/mesh_dsm.tif
[DEBUG]   running /code/build/bin/odm_dem2points -inputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp/mesh_dsm.tif -outputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp/dsm_points.ply -skirtHeightThreshold 1.5 -skirtIncrements 0.2 -skirtHeightCap 100
[DEBUG]   running /code/SuperBuild/src/PoissonRecon/Bin/Linux/PoissonRecon --in /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/tmp/dsm_points.ply --out /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_25dmesh.dirty.ply --depth 9 --pointWeight 4 --samplesPerNode 1.0 --threads 16 --linearFit
[WARNING] Bad average root: 1.000000
(0.425012 0.500634) (0.500634)
[WARNING] Bad average root: -0.000000
(0.500634 0.469191) (0.500634)
[WARNING] Bad average root: -0.000000
(0.500634 0.328197) (0.500634)
[DEBUG]   running /code/build/bin/odm_cleanmesh -inputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_25dmesh.dirty.ply -outputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_25dmesh.ply -removeIslands -decimateMesh 100000
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_texturing/odm_textured_model.obj
[DEBUG]   running /code/SuperBuild/install/bin/texrecon /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/opensfm/reconstruction.nvm /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_meshing/odm_mesh.ply /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none       -n 65535
/code/SuperBuild/install/bin/texrecon (built on Sep 16 2018, 20:40:10)
Load and prepare mesh:
PLY Loader: comment VTK generated PLY File
Reading PLY: 97294 verts...
194459 faces...
done.
Generating texture views:
NVM: Loading file...
NVM: Number of views: 60
NVM: Number of features: 0
Loading 100%... done. (Took 2.421s)
Building adjacency graph:
Adding edges 100%... done. (Took 0.35s)
291611 total edges.
View selection:
Building BVH from 194459 faces...
done. (Took: 190 ms)
Calculating face qualities 100%... done. (Took 1.975s)
Postprocessing face infos 100%... done. (Took 0.113s)
Maximum quality of a face within an image: 2915.27
Clamping qualities to 340.538 within normalization.
Writing data cost file...
done.
Optimizing:
Time[s]	Energy
0	198093
0	197041
0	195445
0	194609
1	194204
1	193973
1	193849
1	193778
1	193711
1	193663
[... 104 lines cut ...]
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running ODM DEM Cell
[INFO]    Classify: False
[INFO]    Create DSM: True
[INFO]    Create DTM: False
[INFO]    DEM input file /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_georeferencing/odm_georeferenced_model.laz found: False
[WARNING] DEM will not be generated
[INFO]    Running ODM DEM Cell - Finished
[INFO]    Running ODM Orthophoto Cell
[WARNING] Cannot calculate GSD, using requested resolution of 5.0
[DEBUG]   running /code/build/bin/odm_orthophoto -inputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_texturing_25d/odm_textured_model.obj -logFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_orthophoto/odm_orthophoto_log.txt -outputFile /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c/odm_orthophoto/odm_orthophoto.png -resolution 20.0  -outputCorne [... cut ...]
[WARNING] No geo-referenced orthophoto created due to missing geo-referencing or corner coordinates.
[INFO]    Running ODM OrthoPhoto Cell - Finished
[INFO]    OpenDroneMap app finished - Thu Sep 27 14:27:17  2018
Postprocessing: /var/www/data/e242b02a-aa88-4e15-a433-debedf517d6c

No orthophoto found at odm_orthophoto/odm_orthophoto.tif: will skip tiling

No orthophoto found at odm_orthophoto/odm_orthophoto.tif: will skip MBTiles generation

Potree point cloud will not be generated (no suitable input files found)

Postprocessing: done (•̀ᴗ•́)و!

Compressing all.zip

pierotofy, 2018-10-23:
Fixed.

closed 2018-10-23 by pierotofy

smathermather, 2018-10-23:
w00t! Thanks.