WebODM/WebODM #533: Add layer to project

ivanmalo, 2018-10-05:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What is the problem?

I would need to add contour lines or georeferenced layers to existing projects.

--

Hi, 

We are looking into adding layer to existing projects, layers generated in Q Gis, but it seem impossible to add them to the project. Is there any way we can achieve this?

pierotofy, 2018-10-05:
Hi @ivanmalo,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-10-08 17:40) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

saricicekibrahim, 2018-12-26:
Hi there,

3 way comes to my mind,
- Adding a wms or tms layer dynamically. May be globally or for the project. And serving own data from a map server (Geoserver, Mapnik, Mapserver, etc..) as base or overlay layer.
- Adding a layer using local shapefile, geojson. May save to postgres and find a way to serve it. May be vector layer with geojson requests.
- Or drag & dropping a zipped shapefile or geojson on to map. Layer will be used only on recent session only. Like this example http://leaflet.calvinmetcalf.com/#3/32.69/10.55
I made a work for the third one;
https://github.com/saricicekibrahim/WebODM/tree/templatelayer

pierotofy, 2019-04-03:
You can also now preview and export contours as part of #639 so this can be closed.

closed 2019-04-03 by pierotofy