WebODM/WebODM #536: Unaccurate DSMs/DTMs from webodm

Peketronics, 2018-10-11:
Hello, I’m running the latest webodm on docker and i`m having problems with DSMs/DTMs generated with webodm, it shows elevations where there are not from eye view on terrain… compared to the same data sets processed with pix4d and drn2map wich look accurate to the tarrain and I’m not being able to succesfully post process the DSMs/DTMs in Qgis… any idea why??? Data sets where captured with a Mavic pro a screenshot from both dsm opened in qgis, please see the differences in the layer attributes section. 
[image]

would it have to do wih photo distortion?? am I supposed to calibrate the photos of the mavic pro before feeding to webodm?? Here’s a screenshot from a exif data viewer of all the photos
[image]
Fligth heigth was 40 meters using pix4d fligth app :slight_smile:

pierotofy, 2018-10-11:
Hi @Peketronics,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-10-14 12:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Peketronics, 2018-10-11:
How did you install WebODM? (Docker, natively, ...)?
DOcker

What's your browser and operating system?
 Chrome 69 on Windows 10

What is the problem?
Explained in first post

What should be the expected behavior?
Output DSMs that are similar to terrain 

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Im using high resolution preset to process

pierotofy, 2018-10-11:
A discussion is already under way at http://community.opendronemap.org/t/problem-with-webodm-generated-dsm-dtm/1209/5

smathermather, 2018-10-13:
Reopening and renaming.

smathermather, 2018-10-13:
@pierotofy -- let me know if you want me to move this to OpenDroneMap/OpenDroneMap, as it is an underlying issue core ODM. Also, it may require work upstream in OpenSfM.

pierotofy, 2018-10-13:
Yeah this should probably be opened in ODM. You can close this one.

smathermather, 2018-10-13:
@Peketronics, can you add your observations to this issue? https://github.com/OpenDroneMap/OpenDroneMap/issues/538

closed 2018-10-13 by smathermather