WebODM/WebODM #538: WebpackError at /welcome/ after using Install Script Ubuntu 16.04

Wlanfr3ak, 2018-10-16:
### How did you install WebODM? (Docker, natively, ...)?

WebODM Script for Ubuntu Server 16..04 LTS

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 62 on Windows 10

### What is the problem?

WebODM Website didnt open after Installing WebODM with the Installer

### What should be the expected behavior?

Normally in my Tests i can now set a Password

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Try to install the Script on a 16.04.5 64Bit

Output: 

WebpackError at /welcome/

            ModuleBuildError in 
            Module build failed (from ./node_modules/extract-text-webpack-plugin/dist/loader.js):
ModuleBuildError: Module build failed (from ./node_modules/sass-loader/lib/loader.js):
Error: Cannot find module 'node-sass'
    at Function.Module._resolveFilename (module.js:548:15)
    at Function.Module._load (module.js:475:25)
    at Module.require (module.js:597:17)
    at require (/usr/lib/node_modules/webpack-cli/node_modules/v8-compile-cache/v8-compile-cache.js:159:20)
    at Object.sassLoader (/webodm/node_modules/sass-loader/lib/loader.js:46:72)
    at runLoaders (/usr/lib/node_modules/webpack/lib/NormalModule.js:286:20)
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:364:11
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:230:18
    at runSyncOrAsync (/usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:143:3)
    at iterateNormalLoaders (/usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:229:2)
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:202:4
    at process.nextTick (/usr/lib/node_modules/webpack/node_modules/enhanced-resolve/lib/CachedInputFileSystem.js:73:15)
    at _combinedTickCallback (internal/process/next_tick.js:132:7)
    at process._tickCallback (internal/process/next_tick.js:181:9)
            

Request Method: 	GET
Request URL: 	http://<IP-Adress>/welcome/
Django Version: 	2.0.3
Exception Type: 	WebpackError
Exception Value: 	

            ModuleBuildError in 
            Module build failed (from ./node_modules/extract-text-webpack-plugin/dist/loader.js):
ModuleBuildError: Module build failed (from ./node_modules/sass-loader/lib/loader.js):
Error: Cannot find module 'node-sass'
    at Function.Module._resolveFilename (module.js:548:15)
    at Function.Module._load (module.js:475:25)
    at Module.require (module.js:597:17)
    at require (/usr/lib/node_modules/webpack-cli/node_modules/v8-compile-cache/v8-compile-cache.js:159:20)
    at Object.sassLoader (/webodm/node_modules/sass-loader/lib/loader.js:46:72)
    at runLoaders (/usr/lib/node_modules/webpack/lib/NormalModule.js:286:20)
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:364:11
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:230:18
    at runSyncOrAsync (/usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:143:3)
    at iterateNormalLoaders (/usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:229:2)
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:202:4
    at process.nextTick (/usr/lib/node_modules/webpack/node_modules/enhanced-resolve/lib/CachedInputFileSystem.js:73:15)
    at _combinedTickCallback (internal/process/next_tick.js:132:7)
    at process._tickCallback (internal/process/next_tick.js:181:9)
            

[... 194 lines cut ...]
  23.     bundle = get_loader(config).get_bundle(bundle_name)

File "/webodm/python3-venv/lib/python3.5/site-packages/webpack_loader/loader.py" in get_bundle
  97.             raise WebpackError(error)

Exception Type: WebpackError at /welcome/
Exception Value: 
            ModuleBuildError in 
            Module build failed (from ./node_modules/extract-text-webpack-plugin/dist/loader.js):
ModuleBuildError: Module build failed (from ./node_modules/sass-loader/lib/loader.js):
Error: Cannot find module 'node-sass'
    at Function.Module._resolveFilename (module.js:548:15)
    at Function.Module._load (module.js:475:25)
    at Module.require (module.js:597:17)
    at require (/usr/lib/node_modules/webpack-cli/node_modules/v8-compile-cache/v8-compile-cache.js:159:20)
    at Object.sassLoader (/webodm/node_modules/sass-loader/lib/loader.js:46:72)
    at runLoaders (/usr/lib/node_modules/webpack/lib/NormalModule.js:286:20)
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:364:11
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:230:18
    at runSyncOrAsync (/usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:143:3)
    at iterateNormalLoaders (/usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:229:2)
    at /usr/lib/node_modules/webpack/node_modules/loader-runner/lib/LoaderRunner.js:202:4
    at process.nextTick (/usr/lib/node_modules/webpack/node_modules/enhanced-resolve/lib/CachedInputFileSystem.js:73:15)
    at _combinedTickCallback (internal/process/next_tick.js:132:7)
    at process._tickCallback (internal/process/next_tick.js:181:9)

Wlanfr3ak, 2018-10-16:
Okay, i test it again and with the solution from the manual it works now.

closed 2018-10-16 by Wlanfr3ak

smathermather, 2018-10-17:
Excellent!