WebODM/WebODM #540: Process exited with code 1 on Ubuntu server

RuikangSun, 2018-10-21:
It is said that this problem is because the memory is rang out. But I checked the server dashboard, most of the memory is available. Server information: 4 vCPU 16 GB

RuikangSun, 2018-10-21:
[image]
The left is CPU usage and the right is memory usage.

pierotofy, 2018-10-21:
Hi @RuikangSun,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-10-23 22:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2018-10-23:
Let's move the conversation over to http://community.opendronemap.org/c/webodm, thank you.

closed 2018-10-23 by pierotofy