WebODM/WebODM #541: docker-compose version is unsupported (ubuntu 16.04)

merkato, 2018-10-22:
### How did you install WebODM? (Docker, natively, ...)?

 By Docker

### What's your browser and operating system? 

Ubuntu 16.04.5, Docker-ce, docker-compose 1.8.0

### What is the problem?

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.plugins.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml -f docker-compose.plugins.yml up
ERROR: Version in "./docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a version of "2" (or "2.0") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/
ERROR: Version in "./docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a version of "2" (or "2.0") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/

### What should be the expected behavior?

Correct installation process.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Probably not so easy reproducible. This process is correct on Ubuntu 18.04 and newer, with docker-compose 1.17

MatthiasSiewert, 2019-05-13:
Had the ecact same issue. This anwser worked for me:
[https://community.opendronemap.org/t/docker-compose-version-is-unsupported/556/6](url)

pierotofy, 2020-04-02:
I don't think this is fixable on our end (it's a docker-compose problem/bug).

closed 2020-04-02 by pierotofy