WebODM/WebODM #543: need at least one array to concatenate

bonaime, 2018-10-26:
### How did you install WebODM? (Docker, natively, ...)?
Docker on virtual Host Ubuntu

### What's your browser and operating system?
Firefox 62 on macOS (Mojave)

### What is the problem?
 "Process exited with code 1" means that part of the processing failed. Try tweaking the Task Options as follows:

    Increase the min-num-features option, especially if your images have lots of vegetation

log is : 

```
12:59:37,473 DEBUG: Undistorting image SiteMbis-drone_09584.jpg
2018-10-26 12:59:37,478 DEBUG: Undistorting image SiteMbis-drone_09591.jpg
2018-10-26 12:59:37,479 DEBUG: Undistorting image SiteMbis-drone_09550.jpg
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /var/www/data/615bcff8-e5ec-4510-87f2-6b4d769302f8/opensfm
2018-10-26 12:59:53,439 INFO: Computing neighbors
2018-10-26 12:59:56,026 INFO: Merging depthmaps
Traceback (most recent call last):
File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
command.run(args)
File "/code/SuperBuild/src/opensfm/opensfm/commands/compute_depthmaps.py", line 25, in run
dense.compute_depthmaps(data, graph, reconstructions[0])
File "/code/SuperBuild/src/opensfm/opensfm/dense.py", line 52, in compute_depthmaps
merge_depthmaps(data, graph, reconstruction, neighbors)
File "/code/SuperBuild/src/opensfm/opensfm/dense.py", line 222, in merge_depthmaps
points = np.concatenate(points)
ValueError:
need at least one array to concatenate
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/run_opensfm.py", line 157, in process
(context.pyopencv_path, context.opensfm_path, tree.opensfm))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

I am using 100 pictures and a gcp_list.txt file.

pierotofy, 2018-10-30:
Hey @bonaime, thanks for reporting the problem. :hand:

It looks like you are either 1) Running an older version of OpenDroneMap or 2) You passed the `--use-opensfm-dense` option. If it's the case of 1) you should try to update to the latest version and try again.

If it's the case of 2) or updating doesn't help, having a copy of the dataset images would help greatly.

pierotofy, 2020-04-02:
Closing due to inactivity.

closed 2020-04-02 by pierotofy