WebODM/WebODM #544: GetProj4StringSPI: Cannot find SRID (9001) in spatial_ref_sys

Sticklord, 2018-10-27:
### How did you install WebODM? (Docker, natively, ...)?

Using Docker Toolbox through Window10 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Window 10 Home-1803
https://whatismybrowser.com/w/P6GQYHT

### What is the problem?

Process got stuck at "READING:  odm_georeferencing/odm_georeferenced_model.laz" in WebODM", i noticed the error kept on repeating in the worker container.

Here's the last part of the log from Docker Quickstart Terminal, let me know if you'd like more.

```worker        | [2018-10-27 21:22:25,240: ERROR/ForkPoolWorker-4] worker.tasks.process_task[82e6c0c4-2549-4918-b9b4-b622fa93ea27]: Uncaught error! This is potentially bad. Please report it to http://github.com/OpenDroneMap/WebODM/issues: GetProj4StringSPI: Cannot find SRID (9001) in spatial_ref_sys
worker        |  Traceback (most recent call last):
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 85, in _execute
worker        |     return self.cursor.execute(sql, params)
worker        | psycopg2.InternalError: GetProj4StringSPI: Cannot find SRID (9001) in spatial_ref_sys
worker        |
worker        |
worker        | The above exception was the direct cause of the following exception:
worker        |
worker        | Traceback (most recent call last):
worker        |   File "/webodm/worker/tasks.py", line 54, in process_task
worker        |     task.process()
worker        |   File "/webodm/app/models/task.py", line 501, in process
worker        |     self.save()
worker        |   File "/webodm/app/models/task.py", line 224, in save
worker        |     super(Task, self).save(*args, **kwargs)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/models/base.py", line 729, in save
worker        |     force_update=force_update, update_fields=update_fields)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/models/base.py", line 759, in save_base
worker        |     updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/models/base.py", line 823, in _save_table
worker        |     forced_update)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/models/base.py", line 872, in _do_update
worker        |     return filtered._update(values) > 0
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/models/query.py", line 709, in _update
worker        |     return query.get_compiler(self.db).execute_sql(CURSOR)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1377, in execute_sql
worker        |     cursor = super().execute_sql(result_type)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1066, in execute_sql
worker        |     cursor.execute(sql, params)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 68, in execute
worker        |     return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 77, in _execute_with_wrappers
worker        |     return executor(sql, params, many, context)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 85, in _execute
worker        |     return self.cursor.execute(sql, params)
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/utils.py", line 89, in __exit__
worker        |     raise dj_exc_value.with_traceback(traceback) from exc_value
worker        |   File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 85, in _execute
worker        |     return self.cursor.execute(sql, params)
worker        | django.db.utils.InternalError: GetProj4StringSPI: Cannot find SRID (9001) in spatial_ref_sys
worker        |
worker        |
worker        | INFO Canceling Task [Task of 2018-10-27T19:23:38.702Z] (ede336bc-9767-48da-93c5-a714d62a300e)
```

### What should be the expected behavior?

Completing the task.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I used the default DSM + DTM options to start the process with GCP's. What I suspect would be of interest is the proj4 string i used in the gcp_list.txt file:

+proj=utm +zone=32 +ellps=GRS80 +towgs84=0,0,0,0,0,0,0 +units=m +vunits=m +no_defs 

Here's a OneDrive link for the set i used: [link](https://1drv.ms/f/s!AuOB8mJEeeBhhh5MzCPBkEINwdtE)

jgrocha, 2019-03-05:
@Sticklord Do you know how to solve this issue? I'm having the same `ERROR:  GetProj4StringSPI: Cannot find SRID (9001) in spatial_ref_sys`in my log.

pierotofy, 2019-04-30:
This is still an issue: https://community.opendronemap.org/t/webodm-stuck-at-post-peocessing/2098/13

pierotofy, 2019-05-01:
This should be caught and an error message be returned instead of freezing the process.

pierotofy, 2019-05-07:
And likely another one: https://community.opendronemap.org/t/post-processing-not-finishing-and-timer-restarts-after-refresh/2152

commit f96b3ea8 (2019-05-23): BUG FIX: #544
closed 2019-05-23 by pierotofy