WebODM/WebODM #547: WebODM sometimes hangs when using a GCP file

pierotofy, 2018-11-02:
http://community.opendronemap.org/t/webodm-seems-to-hang-compressing-all-zip/1112/4

pierotofy, 2018-11-03:
I can't seem to be able to reproduce this on Windows 10 using Docker. Datasets linked in the thread are downloaded successfully.

richardsjogren, 2019-03-05:
I have this problem. If i supply an GCP file, the project hangs on last stage of packing. If i dont have an gcp file, it finishes ok. 
Im really, REALLY, bad att Linux so much questions about that i would have a hard time to answer. 
If you want to know version and other stuff about the installation, please tell me how to do/get that in Ubuntu and i will answer. 
Installed WebODM as "Native" on Ubuntu 16 and have the data on an extra disk.
Also tried the GCP interface by uploading my GCP and then downloading it againg, but stills hangs when i try that file..
I can send a small example with files i that would help.

richardsjogren, 2019-03-06:
Link to [data](https://www.dropbox.com/s/pontkok774qvj30/CantCalcThisWithGCP.zip)

pierotofy, 2019-03-06:
I've been able to process your data on webodm.net

[image]

I'm not sure what would cause it to get stuck. Perhaps low disk space? Memory?

richardsjogren, 2019-03-06:
Ok, thanks, but how do i check that? I have very low skills when it comes to Ubuntu/Docker..sorry.
I have other tasks where i have processed over 600 photos without problem, if i don´t use a GCP-file.
If i process the files i sent without the the GCP i works fine.

Current Diagnostic Information(and as i said i store the results on another disc).
[image]

epurnell, 2019-03-06:
Just a hint on your data collects after viewing your images. 

Industry buildings with linear roof patterns are very hard to process you can see this in your ortho. 

Try to make several collects with different flight path angles to the subject to improve your data. 

Sorry if you are an expirenced Pilot Geospatial data analysis and this was a test 👍🏻

richardsjogren, 2019-03-06:
> Just a hint on your data collects after viewing your images.
> Industry buildings with linear roof patterns are very hard to process you can see this in your ortho.
> Try to make several collects with [...]

Thanks epurnel, i´m not that experienced, but as you mentioned, it´s just some small test data from a larger set. :-)

pierotofy, 2019-03-06:
@richardsjogren when you say that you store the results on another disk, could you expand on that? (how did you install WebODM, did you change any of the default configurations, etc.)

richardsjogren, 2019-03-06:
@pierotofy, well, here we are at my impressive skills about Linux..the first time i followed some guide that installed it. I had same problem with GCP.
The second time, after reinstalling Ubuntu, i used the "native" script.
https://github.com/OpenDroneMap/WebODM/blob/master/contrib/ubuntu_1604_install.sh
After a lot of trial and error it seemed to be working and i used the "--media-dir" to point at my "big disc" for the data. Data is saved to this disc as i should.
https://github.com/OpenDroneMap/WebODM/#where-are-my-files-stored
There was a lot of things i did during the script-installation because the script didnt work, or more likely because im a total noob when it comes to Linux. So i dont have a complete record on how 
 i got it up and running, sorry. :-(

pierotofy, 2019-06-14:
This should now be fixed!

closed 2019-06-14 by pierotofy