WebODM/WebODM #550: Process exited with code 1

RuikangSun, 2018-11-03:
### How did you install WebODM? (Docker, natively, ...)?

sudo apt-get -y update
sudo apt-get -y upgrade
sudo apt-get -y install docker-compose
sudo apt-get -y install python-pip
sudo curl -L https://github.com/docker/compose/releases/download/1.21.0/docker-compose-$(uname -s)-$(uname -m) -o /usr/bin/docker-compose
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input --depth 1
cd WebODM
sudo ./webodm.sh start

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 63 on Windows 7

### What is the problem?

[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/images
[DEBUG]   /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/images
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_4369.JPG | camera: dji fc1102 | dimensions: 3968 x 2976 | focal: 4.49 | ccd: None | lat: 37.0577777778 | lon: 118.094444444 | alt: 50.6
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_4369_sFYkQdR.JPG | camera: dji fc1102 | dimensions: 3968 x 2976 | focal: 4.49 | ccd: None | lat: 37.0577777778 | lon: 118.094444444 | alt: 50.6
[INFO]    Found 2 usable images
[DEBUG]   running /code/build/bin/odm_extract_utm -imagesPath /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/images/ -imageListFile /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/img_list.txt -outputCoordFile /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/odm_georeferencing/coords.txt  -logFile /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/odm_georeferencing/odm_georeferencing_utm_log.txt
[DEBUG]   Line: WGS84 UTM 50N
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   Altitude data detected, enabling it for GPS alignment
[DEBUG]   ['use_exif_size: no', 'feature_process_size: 2048', 'feature_min_frames: 8000', 'processes: 4', 'matching_gps_neighbors: 8', 'depthmap_method: PATCH_MATCH', 'depthmap_resolution: 640', 'depthmap_min_patch_sd: 1', 'depthmap_min_consistent_views: 3', 'optimize_camera_parameters: yes', 'use_altitude_tag: True', 'align_method: naive']
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/opensfm
2018-11-03 14:00:17,008 INFO: Extracting EXIF for DJI_4369.JPG
2018-11-03 14:00:17,189 INFO: Extracting EXIF for DJI_4369_sFYkQdR.JPG
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm detect_features /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/opensfm
2018-11-03 14:00:18,197 INFO: Extracting ROOT_HAHOG features for image DJI_4369.JPG
2018-11-03 14:00:18,472 INFO: Extracting ROOT_HAHOG features for image DJI_4369_sFYkQdR.JPG
2018-11-03 14:00:21,503 DEBUG: Found 10204 points in 2.95629000664s
2018-11-03 14:00:22,155 DEBUG: Found 10204 points in 3.33254504204s
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm match_features /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/opensfm
2018-11-03 14:00:23,544 INFO: Matching 1 image pairs
2018-11-03 14:00:23,929 INFO: Matching DJI_4369_sFYkQdR.JPG  -  1 / 2
2018-11-03 14:00:24,209 INFO: Matching DJI_4369.JPG  -  2 / 2
2018-11-03 14:00:26,473 DEBUG: DJI_4369.JPG - DJI_4369_sFYkQdR.JPG has 10202 candidate matches
2018-11-03 14:00:26,475 DEBUG: Robust matching time : 0.00150108337402s
2018-11-03 14:00:26,475 DEBUG: Full matching 10202 / 10202, time: 2.26585412025s
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm create_tracks /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/opensfm
2018-11-03 14:00:27,101 INFO: reading features
2018-11-03 14:00:27,197 DEBUG: Merging features onto tracks
2018-11-03 14:00:27,326 DEBUG: Good tracks: 10202
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/opensfm
2018-11-03 14:00:28,231 INFO: Starting incremental reconstruction
2018-11-03 14:00:28,403 INFO: 0 partial reconstructions in total.
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/run_opensfm.py", line 141, in process
image_scale = gsd.image_scale_factor(args.orthophoto_resolution, tree.opensfm_reconstruction)
File "/code/opendm/gsd.py", line 31, in image_scale_factor
gsd = opensfm_reconstruction_average_gsd(reconstruction_json)
File "/usr/local/lib/python2.7/dist-packages/repoze/lru/__init__.py", line 348, in cached_wrapper
val = func(*args, **kwargs)
File "/code/opendm/gsd.py", line 75, in opensfm_reconstruction_average_gsd
raise IOError(reconstruction_json + " does not exist.")
IOError: /var/www/data/d581819e-a16e-46a4-9262-f932537e62ee/opensfm/reconstruction.json does not exist.

### What should be the expected behavior?

Not sure

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Not sure

RuikangSun, 2018-11-03:
I tried to update, but it didn't work.

RuikangSun, 2018-11-03:
I tried remove the EXIF header, but it also doesn't work.

[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/images
[DEBUG]   /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/images
[DEBUG]   Loaded DJI_4330.JPG | camera:  | dimensions: 3968 x 2976 | focal: None | ccd: 5.75 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DJI_4329.JPG | camera:  | dimensions: 3968 x 2976 | focal: None | ccd: 5.75 | lat: None | lon: None | alt: None
[INFO]    Found 2 usable images
[DEBUG]   running /code/build/bin/odm_extract_utm -imagesPath /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/images/ -imageListFile /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/img_list.txt -outputCoordFile /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/odm_georeferencing/coords.txt  -logFile /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/odm_georeferencing/odm_georeferencing_utm_log.txt
Error in OdmExtractUtm:
DJI_4330.JPG: No Exif data found in the file
For more detailed information, see log file.
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file
[WARNING] Could not find file /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/odm_georeferencing/coords.txt
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   ['use_exif_size: no', 'feature_process_size: 2048', 'feature_min_frames: 8000', 'processes: 4', 'matching_gps_neighbors: 8', 'depthmap_method: PATCH_MATCH', 'depthmap_resolution: 640', 'depthmap_min_patch_sd: 1', 'depthmap_min_consistent_views: 3', 'optimize_camera_parameters: yes']
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm
2018-11-03 14:16:44,601 INFO: Extracting EXIF for DJI_4330.JPG
2018-11-03 14:16:44,776 INFO: Extracting EXIF for DJI_4329.JPG
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm detect_features /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm
2018-11-03 14:16:45,790 INFO: Extracting ROOT_HAHOG features for image DJI_4330.JPG
2018-11-03 14:16:46,078 INFO: Extracting ROOT_HAHOG features for image DJI_4329.JPG
2018-11-03 14:16:49,111 DEBUG: Found 9788 points in 2.94131302834s
2018-11-03 14:16:49,694 DEBUG: Found 9820 points in 3.2338809967s
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm match_features /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm
2018-11-03 14:16:51,004 WARNING: Not all images have GPS info. Disabling matching_gps_neighbors.
2018-11-03 14:16:51,004 INFO: Matching 1 image pairs
2018-11-03 14:16:51,425 INFO: Matching DJI_4329.JPG  -  1 / 2
2018-11-03 14:16:51,724 INFO: Matching DJI_4330.JPG  -  2 / 2
2018-11-03 14:16:53,345 DEBUG: DJI_4329.JPG - DJI_4330.JPG has 2714 candidate matches
2018-11-03 14:16:53,346 DEBUG: Robust matching time : 0.000809907913208s
2018-11-03 14:16:53,346 DEBUG: Full matching 2677 / 2714, time: 1.920992136s
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm create_tracks /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm
2018-11-03 14:16:53,886 INFO: reading features
2018-11-03 14:16:53,975 DEBUG: Merging features onto tracks
2018-11-03 14:16:54,004 DEBUG: Good tracks: 2677
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm
2018-11-03 14:16:54,548 INFO: Starting incremental reconstruction
2018-11-03 14:16:54,620 INFO: Starting reconstruction with DJI_4329.JPG and DJI_4330.JPG
2018-11-03 14:16:54,780 INFO: Two-view reconstruction inliers: 2674 / 2677
2018-11-03 14:16:54,925 INFO: Triangulated: 1609
2018-11-03 14:16:54,943 DEBUG: Ceres Solver Report: Iterations: 2, Initial cost: 1.148593e+01, Final cost: 9.080192e+00, Termination: CONVERGENCE
2018-11-03 14:16:55,099 DEBUG: Ceres Solver Report: Iterations: 2, Initial cost: 9.537307e+00, Final cost: 8.665784e+00, Termination: CONVERGENCE
2018-11-03 14:16:55,173 DEBUG: Ceres Solver Report: Iterations: 4, Initial cost: 1.542218e+01, Final cost: 1.161922e+01, Termination: CONVERGENCE
2018-11-03 14:16:55,174 INFO: Removed outliers: 0
2018-11-03 14:16:55,186 INFO: -------------------------------------------------------
2018-11-03 14:16:55,232 DEBUG: Ceres Solver Report: Iterations: 0, Initial cost: 1.161922e+01, Final cost: 1.161922e+01, Termination: CONVERGENCE
2018-11-03 14:16:55,232 INFO: Removed outliers: 0
2018-11-03 14:16:55,358 INFO: Reconstruction 0: 2 images, 1610 points
2018-11-03 14:16:55,358 INFO: 1 partial reconstructions in total.
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm export_visualsfm --image_extension png --scale_focal 1 /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm undistort --image_format png --image_scale 1 /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm
2018-11-03 14:16:56,376 DEBUG: Undistorting the reconstruction
2018-11-03 14:16:56,918 DEBUG: Undistorting image DJI_4329.JPG
[... cut ...]
[DEBUG]   running /code/SuperBuild/src/elibs/mve/apps/makescene/makescene /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/opensfm/mve /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/smvs
MVE Makescene (built on Nov  2 2018, 01:41:31)
Info: Detected Noah bundler format.
Reading Bundler file (2 cameras, 1610 features)...
Recognized 2 original images from Noah's Bundler.
Creating output directories...
Saving bundle file...
Writing bundle (2 cameras, 1610 features): /var/www/data/35f22ea4-582e-4dda-a50f-f000f50781d0/smvs/synth_0.out...
Processing view view_0000.mve...

pierotofy, 2018-11-03:
`0 partial reconstructions in total.` indicates that OpenSfM cannot create a reconstruction. Did you capture images with enough overlap and no blurring?

pierotofy, 2018-11-03:
We should probably have a clearer error message for this error in WebODM.

RuikangSun, 2018-11-04:
> 
> 
> `0 partial reconstructions in total.` indicates that OpenSfM cannot create a reconstruction. Did you capture images with enough overlap and no blurring?

Thanks for your reply and updating!
I tried other images with enough overlap and no blurring and it worked! I guess the error is due to the images. A clearer error message for errors is certainly needed.
Best of luck.

closed 2018-11-04 by pierotofy

bonaime, 2018-11-13:
Hi,
I have more or less the same error but my images I think good. 
```
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/images
[DEBUG]   /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/images
[DEBUG]   Loaded DSC09613.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09614.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09610.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09605.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09611.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09612.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09615.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09607.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09609.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09608.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[DEBUG]   Loaded DSC09606.JPG | camera: sony ilce-5100 | dimensions: 6000 x 4000 | focal: 30.0 | ccd: 23.5 | lat: None | lon: None | alt: None
[INFO]    Found 11 usable images
[DEBUG]   running /code/build/bin/odm_extract_utm -imagesPath /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/images/ -imageListFile /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/img_list.txt -outputCoordFile /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/odm_georeferencing/coords.txt  -logFile /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/odm_georeferencing/odm_georeferencing_utm_log.txt
Error in OdmExtractUtm:
Image is missing GPS Latitude data
For more detailed information, see log file.
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file
[WARNING] Could not find file /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/odm_georeferencing/coords.txt
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   ['use_exif_size: no', 'feature_process_size: 2048', 'feature_min_frames: 8000', 'processes: 8', 'matching_gps_neighbors: 8', 'depthmap_method: PATCH_MATCH', 'depthmap_resolution: 640', 'depthmap_min_patch_sd: 1', 'depthmap_min_consistent_views: 3', 'optimize_camera_parameters: yes']
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/opensfm
2018-11-13 12:49:55,417 INFO: Extracting EXIF for DSC09613.JPG
2018-11-13 12:49:55,719 INFO: Extracting EXIF for DSC09614.JPG
2018-11-13 12:49:55,981 INFO: Extracting EXIF for DSC09610.JPG
2018-11-13 12:49:56,260 INFO: Extracting EXIF for DSC09605.JPG
2018-11-13 12:49:56,503 INFO: Extracting EXIF for DSC09611.JPG
2018-11-13 12:49:56,765 INFO: Extracting EXIF for DSC09612.JPG
2018-11-13 12:49:57,024 INFO: Extracting EXIF for DSC09615.JPG
2018-11-13 12:49:57,280 INFO: Extracting EXIF for DSC09607.JPG
2018-11-13 12:49:57,552 INFO: Extracting EXIF for DSC09609.JPG
2018-11-13 12:49:57,831 INFO: Extracting EXIF for DSC09608.JPG
2018-11-13 12:49:58,111 INFO: Extracting EXIF for DSC09606.JPG
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm detect_features /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/opensfm
2018-11-13 12:49:59,079 INFO: Extracting ROOT_HAHOG features for image DSC09613.JPG
2018-11-13 12:49:59,318 INFO: Extracting ROOT_HAHOG features for image DSC09614.JPG
2018-11-13 12:49:59,567 INFO: Extracting ROOT_HAHOG features for image DSC09610.JPG
2018-11-13 12:49:59,767 INFO: Extracting ROOT_HAHOG features for image DSC09605.JPG
2018-11-13 12:49:59,986 INFO: Extracting ROOT_HAHOG features for image DSC09611.JPG
2018-11-13 12:50:00,289 INFO: Extracting ROOT_HAHOG features for image DSC09612.JPG
2018-11-13 12:50:00,527 INFO: Extracting ROOT_HAHOG features for image DSC09615.JPG
2018-11-13 12:50:00,825 INFO: Extracting ROOT_HAHOG features for image DSC09607.JPG
2018-11-13 12:50:03,043 DEBUG: Found 10294 points in 3.32793784142s
2018-11-13 12:50:03,120 DEBUG: Found 10238 points in 3.21973800659s
2018-11-13 12:50:03,527 DEBUG: Found 10255 points in 3.28089404106s
2018-11-13 12:50:03,537 DEBUG: Found 10682 points in 3.16086912155s
2018-11-13 12:50:03,832 DEBUG: Found 10245 points in 3.14990401268s
2018-11-13 12:50:03,892 INFO: Extracting ROOT_HAHOG features for image DSC09609.JPG
2018-11-13 12:50:03,949 INFO: Extracting ROOT_HAHOG features for image DSC09608.JPG
2018-11-13 12:50:04,018 DEBUG: Found 10286 points in 3.09279608727s
2018-11-13 12:50:04,215 DEBUG: Found 10820 points in 3.06835103035s
2018-11-13 12:50:04,366 INFO: Extracting ROOT_HAHOG features for image DSC09606.JPG
[... cut ...]
File "/usr/local/lib/python2.7/dist-packages/repoze/lru/__init__.py", line 348, in cached_wrapper
val = func(*args, **kwargs)
File "/code/opendm/gsd.py", line 75, in opensfm_reconstruction_average_gsd
raise IOError(reconstruction_json + " does not exist.")
IOError
: /var/www/data/1dc93644-e2e9-4f9e-a909-c12a3cbd359a/opensfm/reconstruction.json does not exist.
```

My images are here : 
https://www.dropbox.com/s/gclc6pps6g3eacv/Archive.zip?dl=0

I am using
Firefox 63.0.1 (64 bits)

Webodm is running on a VM with
Ubuntu 7.3.0-16ubuntu3

Thank you