WebODM/WebODM #553: Script hangs on web interface when in task details

xmutantson, 2018-11-06:
### How did you install WebODM? (Docker, natively, ...)?

I used docker and am on windows. I used the manager application.

### What's your browser and operating system? 
Chrome:
https://whatismybrowser.com/w/5NGD9WS

### What is the problem?

Whenever I try to download assets or view either a 3d model or a map, the pages do not load and i get an error that a script is not responding. I also get the error on the first load of the process page. The script is here, https://pastebin.com/3wZ7m4rF and hangs on line 27, which looks like a reactjs script. 

I cannot delete the project either, I am confronted with the same error. As I was typing this, I was going to rerun the process with the same dataset but with default settings, and the interface seems to stop responding as soon as I open task details, even after it loads the process output. I can run new jobs under different processes but I cannot modify anything from a completed job or view anything related to that job.

### What should be the expected behavior?

I expect the buttons to respond within a few minutes and show the data I processed. 

### How can we reproduce this?
All you need to do is to open task details on a completed task. My WebODM Manager is 1.5.7, WebODM itself is 0.6.2 and I have another processing node on a server rack. I updated everything yesterday. 

The dataset options I was using are attached. I know these can be extreme options, but they worked on previous versions and I would expect to be able to at least delete the project. 
[image]

xmutantson, 2018-11-06:
UPDATE: I just restarted all systems completely and the interface, while lagging during loading task output, performed normally afterwards. This is contrary to behavior where it would lag even after the output was loaded. I did restart WebODM through the manager but I guess that something was still open. The computer was also on the same poweron cycle as had been used to update, which may have contributed to the error. All works now.

xmutantson, 2018-11-06:
Ok, I was going to edit the task and attempt to fix some shape distortion, but I noticed that the task doesn't ever load the output after the first view and thus doesn't let you click anything in the interface. I can't restart the whole system every time.

pierotofy, 2018-11-07:
I wonder if it's the same as #483.

As a workaround, have you tried with a different browser such as Firefox?

Thanks for reporting the problem. It's certainly something that needs to be investigated and fixed.

xmutantson, 2018-11-07:
I tried it on edge, firefox, and chrome. Chrome and firefox complained about scripts that were not responding. The page did not load at all on edge but I blame microsoft for that. I do not think it is the same because even when the page loaded the task log after several minutes it was impossible to click anything due to the freezing script. As it is, I cannot use WebODM at all.

pierotofy, 2018-11-08:
Mm, if you go to the Administration panel --> Tasks --> Select the task that is getting stuck, select the "console_output" field and remove all the text from it, then press "Save", does it allow you to expand the task from the dashboard?

Thanks for trying the other browsers. I'm not sure what could cause it to get stuck beside a very long console output.

piachoo, 2018-11-15:
"Administration panel --> Tasks --> Select the task that is getting stuck, select the "console_output" field and remove all the text from it, then press 'Save'" allows to expand the task from the dashboard with no lag.

xmutantson, 2018-11-15:
That did it. The output was 800,000 characters so I see now why that little java thing was freezing. Not going to close the issue because this is still odd behavior, but maybe the console output needs a character counter and a dedicated tab that it opens in so if it's over X amount it will stop it from freezing up.

pierotofy, 2018-12-10:
Fixed with #571

closed 2018-12-10 by pierotofy