WebODM/WebODM #555: Feature Request: File save locations hostMachine/S3

Leafanjing, 2018-11-07:
Currently there is no way to store processed files from WebODM (running in Docker) to either a local computer or Amazon S3 bucket. Either or both options could be very advantageous in terms of ease of use for frequent users.

pierotofy, 2018-11-07:
Hi @Leafanjing,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-11-09 19:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

smathermather, 2018-11-07:
Added the brackets to avoid the GitIssueBot nag... .

frier-sam, 2020-05-22:
Hi ,
Here is what you can try for S3, you can mount S3 storage as fuse file system  (https://medium.com/tensult/aws-how-to-mount-s3-bucket-using-iam-role-on-ec2-linux-instance-ad2afd4513ef)
Then you can give the mounted storage path in .env file for media root. This will save all the images and asserts in S3.
We tried and it works without any problems.

dlebauer, 2021-09-14:
as discussed here: https://community.opendronemap.org/t/cloud-import-on-webodm/2807 adding rclone support would enable read write for over 50 data providers and protocols 

- including: S3, Azure Blob, Nextcloud, ownCloud, OpenStack Swift, SugarSync, Box, Dropbox, Google Drive, OneDrive, SFTP, HTTPS, FTP, WebDAV

See https://rclone.org/

gromain, 2022-02-01:
This is a very interesting feature, one that would definitely add a lot to my processing pipeline.

@dlebauer, has work started on integrating for example rclone?

pierotofy, 2022-02-28:
I'm going to close this as it's possible to save files on the host machine (both via `--media-dir` for docker and by default on native installs).

S3 support is super-tricky as there's no easy way to make the dynamic tiler work with decent performance over S3. Other operations are probably also super-slow on S3 (and we haven't even covered issues of S3's eventual consistency nature).

Still open to add some sort of S3 support (maybe as an import/export plugin?), feel free to open a PR.

closed 2022-02-28 by pierotofy