WebODM/WebODM #556: Can't reset password

posikifi, 2018-11-10:
Running WebODM from docker on Ubuntu 16.04.  Work perfectly few weeks ago. I updated first with "./webodm.sh update"

My browser is latest chrome, but I think it's not relevant here.

I can't reset passwothd with command:

`./webodm.sh resetadminpassword <newpassword`

Response is following:
```
raceback (most recent call last):
  File "manage.py", line 22, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.6/site-packages/django/core/management/__init__.py", line 371, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.6/site-packages/django/core/management/__init__.py", line 365, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.6/site-packages/django/core/management/base.py", line 288, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.6/site-packages/django/core/management/base.py", line 335, in execute
    output = self.handle(*args, **options)
  File "/usr/local/lib/python3.6/site-packages/django/core/management/commands/shell.py", line 92, in handle
    exec(sys.stdin.read())
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/django/db/models/query.py", line 307, in __getitem__
    return qs._result_cache[0]
IndexError: list index out of range

```

pierotofy, 2018-11-10:
Hi @posikifi,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-11-13 08:45) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

posikifi, 2018-11-10:
How did you install WebODM? (Docker, natively, ...)?
Docker

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Browser:
Chromium 70 on Ubuntu Linux

OS:
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial

What is the problem?
Can't change password with ./webodm.sh resetadminpassword

What should be the expected behavior?
Passwod changed

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Hmm,  I don't have other systems here. So I can't reproduce it.

posikifi, 2018-11-10:
Well,

I create AWS Ubuntu 18.04 and it resetpassword work there. Maybe it's my enviroment. Looking more information.

P

pierotofy, 2018-11-11:
Hey @posikifi :hand:

I think that error is caused by the fact that the resetpassword command cannot find any administrator user within the WebODM database. I'm not sure how the database got into that state, but I don't think it's related to the version of Ubuntu.

pierotofy, 2020-12-16:
Closing, since this is probably fixed by now. Feel free to re-open if the issue persists. :pray:

closed 2020-12-16 by pierotofy