WebODM/WebODM #558: SSL TLSV1 Alert

AnthonyCoughlin, 2018-11-11:
Hey Guys,

I've recently reinstalled Docker Toolbox and cloned and updated WebODM.  I'm now receiving the message below. 

Thanks for the help.!

[image]

pierotofy, 2018-11-11:
Hi @AnthonyCoughlin,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2018-11-13 23:35) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

AnthonyCoughlin, 2018-11-11:
How did you install WebODM? (Docker, natively, ...)?
Docker Toolbox and WebODM natively.

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Chrome, Win10

What is the problem?
Receiving an error 'SSL TLSV1 ALERT'

What should be the expected behavior?
WebODM should start

How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Uninstalled and reinstalled Docker Toolbox.  Cloned WebODM and updated.  Ran './webodm.sh start' and received error.

AnthonyCoughlin, 2018-11-11:
I wasn't sure what the error meant, so I also used ipconfig to flush DNS, release, and renew. Still received errror.

AnthonyCoughlin, 2018-11-12:
I tried to update today and received the error on the update as well, see screenshot below.

[image]

AnthonyCoughlin, 2018-11-13:
If I did this correctly, It did not work...

[image]

pierotofy, 2018-11-14:
You can setup environment variables by following this guide: https://www.youtube.com/watch?v=bEroNNzqlF4

AnthonyCoughlin, 2018-11-14:
@pierotofy 

Thanks for the guidance.  Creating the new environment variable (in the correct place) worked.  I'm up and running.

Thanks again for the help!!