WebODM/WebODM #560: Output ODM orthophoto is shifted compared to point cloud and dsm when using GCP

denistestemale, 2018-11-23:
### How did you install WebODM? (Docker, natively, ...)?
Docker on Ubuntu 18.04 + git.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Firefox 63 on Ubuntu Linux.

### What is the problem?
The ouput orthophoto from WebODM is constantly shifted compared to the point cloud and dsm (both PC and dsm are perfectly aligned) when using a gcp_file.txt. This is not the case when no GCP are used. The shift varies depending on the parameters, between 1 and 2m. 

### What should be the expected behavior?
The orthophoto should be aligned with the other assets (.ply PC and dsm.tif are the ones are consider), if it is projected from the PC (if I understand things correctly)

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
The dataset is the same as in: https://github.com/OpenDroneMap/WebODM/issues/549#issuecomment-437840704, you can download it from:
https://drive.google.com/folderview?id=1B1c5BsG19Hx7RYwzUgZ26nZWb702yLbz

I reconstructed this dataset with and without a GCP file. The coordinates in the GCP file are ETRS89/UTM31N (see discussion in the other issue thread) but the same offset shows up if the coordinates are WGS84/UTM31N (or so I think). Based on my understanding (might be crap), the orthophoto and all the other assets are derived from the mesh, after it is georeferenced with the EXIF data or the GCP file. OK, screenshots where I use QGIS and both the measuring tool and the terrain profile tool to demonstrate. 

### if I don't use a GCP file, 
- the dsm and orthophoto are perfectly aligned. We'll take the change of slope as a reference: we can see that the change of slope as visible in the photo is where the abrupt change of altitudes occur in the dsm values at x=28.151. 
[image]

- I also opened the .ply georeferenced point cloud (PC) in CloudCompare et projected a dsm raster along the Z axis. The resulting values are visible in the profile graph above as a red curve, the blue curve being the ODM dsm. They are basically identical (only the resolution is different). Finally I also projected a RGB raster along the Z axis, which is basically the orthophoto (right?), and it is perfectly aligned with ODM orthophoto (I don't show a screenshot). 

### if I use a GCP file
-  I did the same tricks (CloudCompare, projection along Z of a dsm raster and RGB rasters + comparison in QGIS). 
 * first the offset between the odm ortho and CloudCompare projection, about 2m, I observed this number in different tries. 
[image]
 * then in the profile plot the ODM dsm and CloudCompare projected dsm are indistinguishable, 
[image]
 * ... and they are aligned with the CloudCompare orthophoto, but **not** aligned with the ODM dsm as you can see here (the cursor is on the change of slope):
[image]
 * in case you wonder which one is correct, it is definitely the PC (and dsm). 

I'm still skeptical about the SCR thingie (i.e. WGS84UTM31N) and if it is at the origin of the problem (in case it has an impact on how the orthophoto is projected from the PC). But in that case, why is the dsm not projected with the same error/offset? Tell me if I shall try something else. Thanks.
Denis

pierotofy, 2019-02-14:
So I stumbled upon a dataset which exhibited this behavior, and upon further inspection I noticed that the bounding box computed by `odm_orthophoto` is off compared to the values of the bounding box computed with blender.

From blender one can compute the bounding box values by running:

```
>>> obj = bpy.context.active_object
>>> [v[:] for v in obj.bound_box]
```

Here are the results:

```
[(-210.03074645996094, -226.8159942626953, -8.868979454040527), 
(-210.03074645996094, -226.8159942626953, 26.564987182617188), 
(-210.03074645996094, 200.67742919921875, 26.564987182617188), 
(-210.03074645996094, 200.67742919921875, -8.868979454040527), 
(223.90260314941406, -226.8159942626953, -8.868979454040527), 
(223.90260314941406, -226.8159942626953, 26.564987182617188), 
(223.90260314941406, 200.67742919921875, 26.564987182617188), 
(223.90260314941406, 200.67742919921875, -8.868979454040527)]
```

Compared to:

```
Point 1: -204.888 -216.914
Point 2: -204.888 207.406
Point 3: 229.752 207.406
Point 4: 229.752 -216.914
```

The values differ by ~5-7 meters, which is the offset I'm noticing for this particular dataset. I believe different models behave differently (and some models are totally unaffected), which would explain why this bug has been so elusive...

Will look further into this and hopefully have a fix.

denistestemale, 2019-02-14:
Thank you Piero for looking into this. 
Indeed in the same geographic area I had one dataset with not discernable shift, the one I documented here (~10m) and even another one where the bounding box is so shifted that I can only see one half of the orthophoto (over a 10ha surface, i.e. a shift of about 50m if I remember correctly). 

Cheers

pierotofy, 2019-02-15:
Possible fix opened for review: https://github.com/OpenDroneMap/ODM/pull/947

@denistestemale could you try to reprocess with the changes in #947 and report back if this solves the issue?

pierotofy, 2019-02-17:
I've reprocessed this dataset during testing and things seem to be looking good:

[image]

denistestemale, 2019-02-18:
Hi Piero.
From your screenshot it looks indeed very promising. Sorry for not being very reactive. Since you checked on this dataset I'll try on another where the problem was such that the orthophoto was unusable (since it lacked so much in the bounding box). 
I just have to figure out how to update my version of WebODM to a version that includes your modification (but based on [this thread](https://github.com/OpenDroneMap/ODM/pull/947), it will be in the master in the next hours, right?).
I'll also check that it solved the GCP precision issue I always had (and which forced me to use Pix4D), same symptoms as @agroos [here](https://community.opendronemap.org/t/webodm-vs-odm-differences-in-the-precission-and-quality-of-orthophotos-and-dems/1735/6) 

Denis

closed 2019-02-18 by pierotofy