WebODM/WebODM #581: Better plugin management

pierotofy, 2019-01-02:
Current limitations of the plugin system:
 - Plugins can be enabled/disabled without issues only on Unix systems
 - Mounting the /plugins directory as a system volume is a bad idea, because Windows.
 - The command line is not an optimal way to manage plugins. A UI should be provided.
 - A marketplace for plugins should be built
 - The ability to enable/disable/install/uninstall plugins should not require a WebODM restart, like it currently does.

danbjoseph, 2019-07-10:
Plugins can be turned on and off via the admin interface once WebODM is running. Is it possible to pass parameters on install to modify which plugins are turned on from the start?

pierotofy, 2019-07-10:
Yes, you can add a "disabled" file (just an empty file named `disabled`) in the root of a plugin to prevent WebODM from loading it by default. https://github.com/OpenDroneMap/WebODM/blob/master/plugins/elevationmap/disabled

As the number of plugins increases we'll start looking at creating proper plugin repository.

pierotofy, 2020-04-02:
Management for plugins is now implemented in #846  (expect the marketplace).

closed 2020-04-02 by pierotofy