WebODM/WebODM #593: Swap/overwrite existing coordinates in exif with coordinates in a different file

pierotofy, 2019-01-07:
This is useful for PPK workflows.

Possibly related: #282

KD9FNA, 2019-02-28:
I am also looking into solutions to this.  I haven't been able to find a workflow yet that works and unfortunately lack the knowledge to be able to code it.  I am going to be flying an RTK system on my Matrice in the coming months so I should be able to provide example datasets when I get outside.  

My basic setup involves RTKLIB running on a ground station which continuously records the corrected position information separately from the aircraft.  Presumably whatever code is developed for this would have to interpolate the position based on the EXIF information stating when the camera took the photo.

pierotofy, 2021-02-03:
We now have https://docs.opendronemap.org/geo.html

WebODM could provide a UI to import/create these. (Help us?)

closed 2022-12-14 by pierotofy