WebODM/WebODM #595: Installation error (node-odm-1    | /bin/sh: 1: dmidecode: not found)

juacarri, 2019-01-09:
Hello, I'm trying to install WebODM in a Windows machine. After execute the following code:

`git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input --depth 1
cd WebODM
./webodm.sh start`
I'm getting the error **/bin/sh: 1: dmidecode: not found**, after get it the installation program is repeating it and the web is not working in http://localhost:8000

Following I send all the information gave by the console:

> $ ./webodm.sh start
> Checking for docker...   OK
> Checking for git...   OK
> Checking for python...   OK
> Checking for pip...   OK
> Checking for docker-compose...   OK
> Starting WebODM...
> 
> Using the following environment:
>
> Host: localhost
> Port: 8000
> Media directory: appmedia
> SSL: NO
> SSL key:
> SSL certificate:
> SSL insecure port redirect: 80
> Celery Broker: redis://broker
> 
> Make sure to issue a ./webodm.sh down if you decide to change the environment.
> 
> docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml start || docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
> Starting db         ... done
> Starting broker     ... done
> Starting node-odm-1 ... done
> Starting worker     ... done
> Starting webapp     ... done
> ERROR: No containers to start
> Creating network "webodm_default" with the default driver
> Creating broker     ... done
> Creating db         ... done
> Creating node-odm-1 ... done
> Creating worker     ... done
> Creating webapp     ... done
> Attaching to broker, node-odm-1, db, worker, webapp
> broker        | 1:C 09 Jan 2019 10:06:04.237 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
> broker        | 1:C 09 Jan 2019 10:06:04.237 # Redis version=5.0.3, bits=64, commit=00000000, modified=0, pid=1, just started
> broker        | 1:C 09 Jan 2019 10:06:04.237 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
> broker        | 1:M 09 Jan 2019 10:06:04.241 * Running mode=standalone, port=6379.
> broker        | 1:M 09 Jan 2019 10:06:04.242 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
> db            | LOG:  database system was shut down at 2019-01-09 10:05:06 UTC
> broker        | 1:M 09 Jan 2019 10:06:04.242 # Server initialized
> worker        | Postgres is up - executing command
> broker        | 1:M 09 Jan 2019 10:06:04.242 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
> db            | LOG:  MultiXact member wraparound protections are now enabled
> worker        | wait-for-it.sh: waiting for broker:6379 without a timeout
> worker        | wait-for-it.sh: broker:6379 is available after 0 seconds
> db            | LOG:  database system is ready to accept connections
> webapp        | Postgres is up - executing command
> webapp        | wait-for-it.sh: waiting for broker:6379 without a timeout
> webapp        | wait-for-it.sh: broker:6379 is available after 0 seconds
> webapp        |
> webapp        |  _       __     __    ____  ____  __  ___
> webapp        | | |     / /__  / /_  / __ \/ __ \/  |/  /
> webapp        | | | /| / / _ \/ __ \/ / / / / / / /|_/ /
> webapp        | | |/ |/ /  __/ /_/ / /_/ / /_/ / /  / /
> webapp        | |__/|__/\___/_.___/\____/_____/_/  /_/
> webapp        |
> webapp        |
> webapp        | Checking python version... 3.x, good!
> webapp        | Checking GDAL version... GDAL 2.3.1, released 2018/06/22, excellent!
[... 1432 lines cut ...]
> webapp        |
> webapp        | If there are no errors, WebODM should be up and running!
> webapp        |
> webapp        | Open a web browser and navigate to http://localhost:8000
> webapp        |
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> broker        | 1:M 09 Jan 2019 10:06:04.242 * Ready to accept connections
> broker        | 1:M 09 Jan 2019 10:11:05.089 * 100 changes in 300 seconds. Saving...
> broker        | 1:M 09 Jan 2019 10:11:05.091 * Background saving started by pid 14
> broker        | 14:C 09 Jan 2019 10:11:05.092 * DB saved on disk
> broker        | 14:C 09 Jan 2019 10:11:05.092 * RDB: 0 MB of memory used by copy-on-write
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found
> node-odm-1    | /bin/sh: 1: dmidecode: not found

Someone had this issue previously?

pierotofy, 2019-01-09:
Hi @juacarri,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-01-12 05:15) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2019-01-15:
Hi @juacarri :hand: the dmidecode error should be now gone in the latest version, but it shouldn't have affected the availability of WebODM. Maybe instead of localhost you needed to specify the IP address of your machine?

closed 2019-01-15 by pierotofy