WebODM/WebODM #598: native install with splitmerge-ecto fails

danbjoseph, 2019-01-13:
ran `bash configure.sh install` on AWS EC@ - Ubuntu Server 16.04 LTS (HVM), SSD Volume Type - ami-076e276d85f524150 (64-bit x86) with `splitmerge-ecto`  https://github.com/OpenDroneMap/ODM/commit/3eac0c7048940e815b3fa7a2d7d9f1434295f9ba

and got `-- Configuring incomplete, errors occurred!`
- [install-output-log.txt](https://github.com/OpenDroneMap/WebODM/files/2753530/install-output-log.txt)
- [CMakeError.log](https://github.com/OpenDroneMap/WebODM/files/2753532/CMakeError.log)
- [CMakeOutput.log](https://github.com/OpenDroneMap/WebODM/files/2753533/CMakeOutput.log)

pierotofy, 2019-01-13:
Hi @danbjoseph,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-01-16 16:00) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

danbjoseph, 2019-01-13:
apologies. logged this on the wrong repository.

closed 2019-01-13 by danbjoseph

smathermather, 2019-01-13:
It’s easy to tell as Piero has his bots managing the ones he’s started... :).