WebODM/WebODM #601: Password error

AnthonyCoughlin, 2019-01-18:
### How did you install WebODM? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Win 10, Chrome

### What is the problem?

Incorrect password error, similar to (#556)

### What should be the expected behavior?

I have not changed the password from that of the default.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I updated on 18Jan19 WebODM and the issue began.  Since then I have uninstalled docker and its components, removed the WebODM dir, and reinstalled all.  Still experiencing the issue.  I've checked and tried the troubleshooting tips, I've also read some of the material on the issues page.  Just need some guidance.  See screenshots below.

[image]

[image]

[image]

AnthonyCoughlin, 2019-01-20:
Ok, removed WebODM again and reran the git and updated.  I used the 'docker ps' code to get the hash (#218).  
[image]

I used the first bash number in the list to use in the code recommended in #218 and got this error.
[image]

I'm still stuck over here....

AnthonyCoughlin, 2019-01-23:
Accidental breakthrough...

This afternoon I deleted all the files\folders on my machine associated with docker and webodm.  Reinstalled docker and cloned webodm.  Still getting the invalid credentials error.  I'm able to retrieve the bash from the webapp but when I try to set the user and pass is still get the same error.  I looked at the code provided in #218 and then #224.  I ended up testing a few variations of the code.  I changed 'is_superuser=True' to "False" and got back a username of 'AnonymousUser' (which I did not create), see the code below:

$ docker exec -ti 170528c20cbe bash -c "echo \"from django.contrib.auth.models import User;print(User.objects.filter(is_superuser=False).first().username)\" | p
ython manage.py shell"

AnonymousUser

After that I replaced 'admin' with "AnonymousUser' in the code you provided in #218 to set the username and pass to 'admin' and I am able to log in now.

image.png

Please help me get things back to normal, because I feel super-unstable over here now!  Haha.  I hope this helps you diagnose the issue.  Please let me know if you need more details.

pierotofy, 2019-01-29:
Thanks for reporting your progress @AnthonyCoughlin ! I will close this for now as it seems like it might have been a hiccup with docker. Feel free to re-open it if the error happens again.

closed 2019-01-29 by pierotofy

frntline-dev, 2023-05-23:
@AnthonyCoughlin Thanks, your solution worked.

All I had to do was click the logo on the top left and then the confetti appeared.

AnthonyCoughlin, 2023-05-24:
@frntline-dev

Glad to hear it helped!  …when you see that confetti…!!!!