WebODM/WebODM #603: The example for creating new task

shoudaos, 2019-01-21:
Hi. Can you give a working example of how to create a new task?

I want to create new tasks in WebODM from nodejs server. So i have urls with images, i download them into my nodejs server and then create POST request to webodm api. And it returns only "Bad request". And I don't even know how to figure out it, how to debug your django API or something else, cause i'm not a python developer.

Maybe if you provide some examaple, how to upload images from nodejs server, it helps me

Thank you.

pierotofy, 2019-01-21:
Hi @shoudaos,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2019-01-24 08:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

shoudaos, 2019-01-21:
@pierotofy yes, sorry about that, i will ask it in the forum

closed 2019-01-21 by shoudaos