WebODM/WebODM #604: [New Feature] Make WebODM Projects more Customer Focused

epurnell, 2019-01-22:
Hi Piero Ticket Opened as Requested

Regarding project interface here are some suggestions to make the project interface more useful for a drone operator. 

Additional optional meta data for the project to make projects easier to catalog:

- Customer Name 
- Customer ID 
- Further Customer Fields (address, phone email, tax number etc etc) 
- RPAS system used 
- Sensor used 
- Collect Location (could be populated by the GPS data) 
- Collect Date and Time 
- Ground station used 
- Contextual images 

Sometimes when I do a survey job I like to capture some single shots that ‘set the scene’ but I don’t process them. It would be good to have the option to add a few images that will not be processed but display under the project . This could make a thumbnail for the project as well. 

Making projects more customer focused rather than project will make a huge difference. For example when I want to recall surveys over time I just want to see all surveys for ‘joe bloggs & co’. Not have to find each project among others.

smathermather, 2019-01-22:
Love this.

pierotofy, 2019-02-22:
This is very much a good feature. It involves changing quite a bit of the "core" ideas about the current dashboard interface, so it needs to be carefully planned from a user interface perspective.

I need to keep thinking about this. Another concept to integrate has been mentioned in #630, so I'm starting to get a clearer picture of a better workflow. In practice we are expanding the scope of the program to go from simple processing to a fuller platform for aerial data management.

epurnell, 2019-02-22:
Hi Piero yes agree it’s fits with the philosophy of #630 in that data management for a drone operator is in some respects as important as data processing itself. Feel free to reach out to me if you would like anymore feedback on the refining the workflow I’m reachable on Skype and you have my email 👍🏻.

joecoc1916, 2020-06-08:
I myself have heard from Surveyors in my geographical area (Florida) that they often take images that need no further processing.

That type workflow should be considered and provided for.

Joseph O'Connor