WebODM/WebODM #607: Change THEME setting does not load

sann1432, 2019-01-25:
Theme settings unavailable in all deployment versions (docker & Native)

Internal Server Error - page does not load 

****************************************

### How did you install WebODM? (Docker, natively, ...)?

Docker, natively, - installed on windows & Linux and have tried docker as well as native installs. 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

https://whatismybrowser.com/w/F8FGCAC

### What is the problem?

unable to change theme setting 

### What should be the expected behavior?

the theme setting should open and allow me to change the settings 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

pull the current code from git into a fresh install on any systems and you will reproduce the issue

pierotofy, 2019-01-25:
Thanks for reporting the issue @sann1432 ! This is clearly a software fault. Hopefully we'll have a fix in a bit. :pray:

sann1432, 2019-01-26:
cheers mate

PR #610 (2019-01-26): Fixed theme settings page, added tests
Fixes #607
closed 2019-01-26 by pierotofy