WebODM/WebODM #608: Process Exited code 1

ivanmalo, 2019-01-25:
### How did you install WebODM? (Docker, natively, ...)?

Natively with commercial script.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu 16.04 and Chrome

### What is the problem?

My WebODM used to work great, but around a week ago, I can't process any images. I have even made new installations and I have the same issue. I'm using the same picture set as the one I had been using and working. I've used many others too and still no luck.

### What should be the expected behavior?

Be able to process images.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Install WebODM with native installe -> start ODM -> Process Images

Console:

[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/images
[DEBUG]   /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/images
[DEBUG]   Loaded DJI_0174.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5103174444 | lon: -100.324579833 | alt: 2235.086
[DEBUG]   Loaded DJI_0175.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5101502778 | lon: -100.324511306 | alt: 2235.086
[DEBUG]   Loaded DJI_0180.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5101284444 | lon: -100.324866833 | alt: 2235.286
[DEBUG]   Loaded DJI_0173.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5104292778 | lon: -100.324625722 | alt: 2235.086
[DEBUG]   Loaded DJI_0179.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5099547778 | lon: -100.324798222 | alt: 2235.286
[DEBUG]   Loaded DJI_0178.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5097916667 | lon: -100.324738806 | alt: 2235.286
[DEBUG]   Loaded DJI_0177.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5096746667 | lon: -100.324701833 | alt: 2235.186
[DEBUG]   Loaded DJI_0176.JPG | camera: dji fc330 | dimensions: 2048 x 1536 | focal: 3.61 | ccd: 6.25 | lat: 20.5100413333 | lon: -100.324468417 | alt: 2235.086
[INFO]    Wrote images database: /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/images.json
[INFO]    Found 8 usable images
[DEBUG]   running /code/build/bin/odm_extract_utm -imagesPath /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/images/ -imageListFile /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/img_list.txt -outputCoordFile /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/odm_georeferencing/coords.txt  -logFile /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/odm_georeferencing/odm_georeferencing_utm_log.txt
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
Warning: Directory Photo has an unexpected next pointer; ignored.
Warning: Directory Iop has an unexpected next pointer; ignored.
Warning: Directory GPSInfo has an unexpected next pointer; ignored.
[DEBUG]   Line: WGS84 UTM 14N
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   Altitude data detected, enabling it for GPS alignment
[DEBUG]   ['use_exif_size: no', 'feature_process_size: 2048', 'feature_min_frames: 50000', 'processes: 8', 'matching_gps_neighbors: 8', 'depthmap_method: PATCH_MATCH', 'depthmap_resolution: 640', 'depthmap_min_patch_sd: 1', 'depthmap_min_consistent_views: 3', 'optimize_camera_parameters: yes', 'use_altitude_tag: True', 'align_method: naive']
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /www/data/877f0c05-8533-47da-9107-dc1d8fba8c56/opensfm
2019-01-25 13:08:08,887 INFO: Extracting EXIF for DJI_0174.JPG
2019-01-25 13:08:08,958 INFO: Extracting EXIF for DJI_0175.JPG
2019-01-25 13:08:09,024 INFO: Extracting EXIF for DJI_0180.JPG
2019-01-25 13:08:09,090 INFO: Extracting EXIF for DJI_0173.JPG
[... 44 lines cut ...]
return self.fn(*self.args, **self.kwargs)
File "/usr/local/lib/python2.7/dist-packages/loky/process_executor.py", line 337, in _process_chunk
return [fn(*args) for args in chunk]
File "/code/SuperBuild/src/opensfm/opensfm/commands/detect_features.py", line 78, in detect
data.save_features(image, p_sorted, f_sorted, c_sorted)
File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 422, in save_features
self._save_features(self._feature_file(image), image, points, descriptors, colors)
File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 407, in _save_features
colors=colors)
File "/usr/local/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 691, in savez_compressed
_savez(file, args, kwds, True)
File "/usr/local/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 700, in _savez
file = os_fspath(file)
File "/usr/local/lib/python2.7/dist-packages/numpy/compat/py3k.py", line 237, in os_fspath
"not " + path_type.__name__)
TypeError: expected str, bytes or os.PathLike object, not unicode
"""
Traceback (most recent call last):
File "/code/run.py", line 47, in <module>
plasm.execute(niter=1)
File "/code/scripts/run_opensfm.py", line 115, in process
(context.pyopencv_path, context.opensfm_path, tree.opensfm))
File "/code/opendm/system.py", line 34, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

pierotofy, 2019-01-25:
Hey thanks for reporting the problem @ivanmalo This was actually caused by a bug introduced by `numpy` (a library we use) and was fixed in https://github.com/OpenDroneMap/ODM/pull/936.

For your purpose, to fix ODM simply issue the following from the shell:

```bash
# sudo pip install numpy==1.15.4
```

ivanmalo, 2019-01-26:
Thanks 

> Hey thanks for reporting the problem @ivanmalo This was actually caused by a bug introduced by `numpy` (a library we use) and was fixed in [OpenDroneMap/ODM#936](https://github.com/OpenDroneMap/ODM/pull/936).
> 
> For your purpose, to fix ODM simply issue the following from the shell:
> 
> ```shell
> # sudo pip install numpy==1.15.4
> ```

Thanks @pierotofy but I'm still getting an error:

Traceback (most recent call last):
File "/code/run.py", line 11, in <module>
from scripts.odm_app import ODMApp
File "/code/scripts/odm_app.py", line 13, in <module>
from odm_meshing import ODMeshingCell
File "/code/scripts/odm_meshing.py", line 7, in <module>
from opendm import mesh
File "/code/opendm/mesh.py", line 3, in <module>
from gippy import GeoImage
File "/usr/local/lib/python2.7/dist-packages/gippy/__init__.py", line 26, in <module>
from .gippy import init, DataType, GeoImage, GeoVector, Options
File "/usr/local/lib/python2.7/dist-packages/gippy/gippy.py", line 32, in <module>
_gippy = swig_import_helper()
File "/usr/local/lib/python2.7/dist-packages/gippy/gippy.py", line 28, in swig_import_helper
_mod = imp.load_module('_gippy', fp, pathname, description)
ImportError: numpy.core.multiarray failed to import

pierotofy, 2019-01-26:
Mm, try the following:

```bash
cd /code
sudo su odm
git pull origin master
exit
sudo bash configure.sh reinstall
sudo chown -R odm:odm /code
```

The configure.sh command will take a while.

ivanmalo, 2019-01-28:
> bash configure.sh reinstall

Still no luck, I got the following error. I think I will try to reinstall.

```
File "/code/run.py", line 8, in <module>
import ecto
ImportError: No module named ecto
```

ivanmalo, 2019-01-29:
> Could you try to issue the following?
> 
> ```
> pip install -U numpy==1.15.4 https://github.com/OpenDroneMap/gippy/archive/numpyfix.zip
> ```
> Then see if processing works without errors.

First of all, thanks for your quick and awesome help.

It did work, I processed 47 images with only **one issue**, the layer goes to the ocean, I processed a couple of others from other flights and it always throws it to the ocean, the all seem to go to: 

> Bounds: [-102.5487682395128, -69.6561717900024, -102.54528553384446, -69.65519746140026].

 I have an older installation locally where I processed the same flights and were referenced correctly.

[image]

pierotofy, 2019-01-29:
Thanks @ivanmalo, this sounds like this other issue https://github.com/OpenDroneMap/ODM/pull/941 that we also just fixed. Simply issue:

```
cd /code
sudo su odm
git pull origin master
```

Sorry for all of these quirks, this week has been full of bug hunting. Please let me know if you find other issues.

ivanmalo, 2019-01-29:
It's all working correctly, thank you.

closed 2019-01-29 by ivanmalo