WebODM/WebODM #609: [Feature] Ability to upload previous processed data

sann1432, 2019-01-26:
it would be good if we could upload previously processed data from other WebODM instances.

this would mean that data could be stored locally and only uploaded when customers/users needed to see that data.

Saving space on cloud servers 

Thoughts?

nixxet, 2019-02-19:
I would be nice to be able to duplicate tasks without having to upload images again to be able to test other processing methods and compare which would be most effective and have the best output.

pierotofy, 2019-02-20:
Thanks @nixxet, that would be an interesting feature. Could you open a new feature request since it's separate from this one?

nixxet, 2019-02-20:
will do!

epurnell, 2019-02-20:
+1 this a desperately needed. When running on a server the resources consumed by repeatedly uploading the images is huge. There is not an easy way of managing them once loaded either. If this could be in the March sprint that would be a huge bonus.

pierotofy, 2019-02-21:
Hey @epurnell :hand:, that's an interesting idea. It should probably get its own feature request. Perhaps you could open one?

I will close this one as this has been implemented.

[image]

closed 2019-02-21 by pierotofy

epurnell, 2019-02-22:
New request raised #630 thanks Piero 👍🏻